microsoft/AirSim #823: AirSim/Unreal causes system shutdown after some time

omsrisagar, 2018-02-16:
Here are the steps I did:

1. Launch epic games launcher.
2. Launch the default Blocks project which will open the Blocks environment in Unreal Engine.
3. Press the Play button and choose rotor simulation.
4. Quadrotor is shown and I am able to control it using the hello_drone example. Everything seems to work. I can get help menu by pressing F1 and such...
5. I leave it like that for about 4 min, then the Windows shuts down abruptly by turning off the display first.

I have also launched the Unreal Engine via Microsoft Visual Studio 2015 Update 3 using Blocks.sln project which shows memory and CPU usage in the sidebar. I noticed it takes only about 1GB memory and not much CPU usage. So I think it is not due to resource limitations.

System details:
Windows 8.1, 8 GB RAM, i7, Unreal 14.6.3, AMD Radeon HD 6670, OpenGL 4.4, DirectX 9.0c (June 2010, 11.0)

sytelus, 2018-04-12:
Is this consistent? Do you just leave it for 4 min right after play button? We can't reproduce this however if you can consistently then please feel free to reopen.

closed 2018-04-12 by sytelus

omsrisagar, 2018-04-12:
@sytelus Yes this is observed consistently in my case. However, I forgot to mention in the reproducible steps that I am using two monitors, each  with the following specifications - Dell 2410, 1920x1200 resolution. 

I am able to solve this problem by simplying two things:
1. Remove one monitor. (this is major)
2. Rather than generating the Blocks - Unreal Editor window from Visual Studio Blocks project by pressing F5 (Local Windows Debugger), I open UE4Editor.exe separately and then attach the Blocks project debugger to UE4Editor.exe. (I'm not sure how much this contributes to solving this problem).

When I do both of the above things, I am not facing that problem. Seems like it is due to resource limitations only (my PC has only 8 GB RAM). Because of 1 above, I guess memory usage is reduced.

I am posting here for the benefit of others who might encounter the same problem.

omsrisagar, 2018-04-14:
Seems like only 1. solves the issue.