microsoft/AirSim #824: About Q matrix and

yumianhuli1, 2018-02-16:
1、Q – Output  disparity-to-depth mapping matrix by stereoRectify(...)
a.Hello,where is the dispartiy mode in AirSim and how to use it in play mode?
b.How are U get Q Matrix by using camera calibration in unreal4?
c.Howt to get camera parameters like focal length、baseline?Can you make a picture to explain it?

2、The CarMode of AirSim is using unreal4 default/origin vehicle component?How can I get yawrate and some other car parameters in AirSim?

Very Thanks a lot!

sytelus, 2018-02-17:
1 (a) - use settings.json to set subwindow view to disparity. Search for docs.
1 (b) - There is no camera calibration and Q matrix. It's ground truth disparity.
1 (c) - FOV=90, baseline =20cm. This is fixed for now.

2 - Yes, its using advanced vehicle template and PhysX. Look in to Content folder of AirSim plugin.

Please read the docs and search the issues before posting new issues.

yumianhuli1, 2018-02-23:
Hello,
1 I do not want to get FOV=90,I want to konw the focal length which between camera and viewport.
2 I also want to know the viewport is nearplane of camera?if not, how can I get the distance between camera and view point(focal length)?
Thank U!

sytelus, 2018-03-04:
[image]

[source](https://www.cc.gatech.edu/~afb/classes/CS4495-Fall2013/slides/CS4495-05-CameraModel.pdf)

closed 2019-01-22 by msb336