microsoft/AirSim #825: build.cmd command problem and cmake not found

gitstar12, 2018-02-16:
When I followed the process of build AirSim, There were some problems.
put the text build.cmd on command line, I got the message like this.

I am using VS2015 and C++ is installed and cmake v10
##_-------------------------------------------------------------------------------

C:\Users\kbs\source\AirSim>build.cmd
Found cmake version: 3.10.2
Starting cmake to build rpclib...
-- Selecting Windows SDK version to target Windows 10.0.16299.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/Users/kbs/source/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/kbs/source/AirSim/external/rpclib/rpclib-2.2.1/build/CMakeFiles/CMakeError.log".
Microsoft (R) Build Engine Version 14.0.25420.1]
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: There are no project files.
Swtich: ALL_BUILD.vcxproj
Microsoft (R) Build Engine Version 14.0.25420.1]
Copyright (C) Microsoft Corporation. All rights reserved.

MSBUILD : error MSB1009: There are no project files.
Switch: ALL_BUILD.vcxproj

Build failed - see messages above.

##_-------------------------------------------------------------------------------

sytelus, 2018-02-17:
Are you running from VS2015 x64 Native command prompt? When you type cmake --version, what do you see? My guess is that you have V2015 but you haven't installed VC++ components. So C++ compiler is missing and cmake is giving above error.

gitstar12, 2018-02-20:
Thank you sir, the problem is solved.
Sir I am working on a gesture control copter project using an Arduino....is it possible to implement it on airsim??

closed 2018-02-27 by sytelus