microsoft/AirSim #826: How to add a camera

josheligoldman, 2018-02-16:
The Image API page
https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md
doesn't explain how to add cameras to the car. How do I add cameras

sytelus, 2018-02-17:
it's fairly easy if you can deal with bit of C++:

1. First add new base that would hold the camera ([see here](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp#L125)).

2. Mount the camera on the base ([see here](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp#L192)).

It would be nice to drive this from json config. If you have cycles, please feel free to contribute!

olivia-skydio, 2018-02-20:
Would it be possible to add more cameras to the drone @sytelus?

sytelus, 2018-03-04:
Yes, you can add as many cameras you want. We already have 5. Just curious why need more...

closed 2018-04-12 by sytelus