microsoft/AirSim #829: multiple quadrotors in same scene

funwey, 2018-02-18:
I am working on a project that uses two or more quadrotors flying in chain-like formation, one following another.  My questions is How to create more quadrotors within the same level in AirSim? How to use the APIs to control each quadrotor separately?

akshaysw, 2018-02-19:
@funwey this may help you #92 #643 #230 refer this [doc](https://github.com/Microsoft/AirSim/blob/master/docs/multi_vehicle.md)

saihv, 2018-02-19:
Instantiating multiple drones is as simple as dragging more flying pawns into the scene. As far as control goes, you'll have to set separate API endpoints up in your control code. See https://github.com/Microsoft/AirSim/pull/706

saihv, 2018-02-19:
Also https://github.com/Microsoft/AirSim/issues/664 for a sample control script.

kshitijgoel007, 2018-05-04:
Hi! Thank you so much for enabling multi-agent support. Quick question: can we do multiple drones in CV mode?

saihv, 2018-05-04:
Yes, I've had no trouble instantiating multiple drones and accessing their camera feeds in Computer Vision mode.

zhuboshun2015, 2019-05-05:
@saihv I read the video posted on your website. It's great! I have a question: how do you adjust the camera so that it can see all the drones from the air?

msb336, 2019-10-02:
For now there is no camera that automatically encapsulates every drone in the scene. You will have to use the manual camera by pressing `M` and follow the drones using your keyboard.

closed 2019-10-02 by msb336

saihv, 2019-10-09:
> @saihv I read the video posted on your website. It's great! I have a question: how do you adjust the camera so that it can see all the drones from the air?

I just used the manual camera ('M') and adjusted the position to a viewpoint such that all the drones are visible.