microsoft/AirSim #832: using airsim without visualisation

Jacquesval, 2018-02-20:
Hello,
I want to use some reinforcement learning for self driving car algorithm, and I'm wondering if it's possible to use the API without visualization? I just need the sensor data and the collision state.
Thanks

tejaskhot, 2018-02-22:
@sytelus ?

sytelus, 2018-02-27:
@aburgm has added PR https://github.com/Microsoft/AirSim/pull/850. It's very close to be done.

aburgm, 2018-02-27:
Note however that also with #850 it will still open a window, that window will just remain black (or just keep showing the last rendered frame), except for the log messages. So depending on for what exactly you need the visualization disabled, this might or might not be what you are looking for. It's mostly about improving performance of the API or subwindows.

mitchellspryn, 2018-02-28:
BTW, I've run AirSim in Azure via Batch. In that case, AirSim gets started in Session 0 because there is no monitor attached to to computer (although you will need to use the OpenGL3 renderer instead of the DirectX). There is no window created, but the exe is running and the API works.

Jacquesval, 2018-03-01:
Thank you very much it's what I wanted to know. And if I use it on Azure, should I build the code or can I use directly the binaries ?

mitchellspryn, 2018-03-02:
Either should work. From the perspective of Batch, it's just kicking off an arbitrary command, so it shouldn't matter if you're using the exe from the repo or if you're running an exe you've created by integrating AirSim into your custom environment.

sytelus, 2018-04-12:
PR https://github.com/Microsoft/AirSim/pull/850 is merged which allows to turn off rendering for the main view port.

closed 2018-04-12 by sytelus