microsoft/AirSim #837: How to connect AirSim and Steering Wheel(Logitech G27)

Kim-BongSu, 2018-02-22:
Hi. I downloaded the binary version of AirSimNH in windows 10. I want to connect my steering wheel Logitech G27 with AirSim. I don't know how to do that.  Should I change the source code of the binary version?

mitchellspryn, 2018-02-28:
Some steering wheels work out of the box. For those that don't, if it can use the windows gamepad driver, you might have luck using [x360ce](https://www.x360ce.com/) to allow it to emulate a gamepad.

zhangjun-xyz, 2018-03-01:
I found that a [plugin](https://github.com/Microsoft/AirSim/tree/master/Unreal/Plugins/AirSim) added the support of [Logitech G920](https://www.logitechg.com/en-gb/product/g920-driving-force). I wonder whether it is supported out of box.

sytelus, 2018-04-12:
Yes, there is build in support now for G920. We haven't tested other wheels.

See https://github.com/Microsoft/AirSim/issues/897

closed 2018-04-12 by sytelus

hrishikeshtawade04, 2018-09-06:
I have Logitech G920 steering wheel and tried connecting it to the blocks environment. I edited the settings.json file and now it looks like this.
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
    "PhysXCar" : {
	  "RC": {
		"RemoteControlID": 0
	  }
    }
  
}

But when I stop and play the unreal engine I am still getting the  control mode as keyboard. Is there something else to be done?