microsoft/AirSim #838: How to make a custom drone?

Oskiboy, 2018-02-22:
Hi, I wanted to know if it is possible to make a custom drone with AirSim?

The AirSim drone is larger than ours, its lighter, it has a different center of mass, and a different thrust.

Is there some parameters where we can just write in these values so that the built-in drone behaves with updated physics? 

We also need to change the UnrealEngine drone model with our own 3D mesh. Where can we do this?

Sincerely

sytelus, 2018-04-12:
You can create 3D mesh and pawn BP and use this setting to replace default one: https://github.com/Microsoft/AirSim/blob/v1.1.10/docs/settings.md#pawnpaths

You can create new config like this for physics parameters: https://github.com/Microsoft/AirSim/blob/v1.1.10/AirLib/include/vehicles/multirotor/configs/SimpleFlightQuadX.hpp

closed 2018-04-12 by sytelus

AloshkaD, 2018-12-21:
@Oskiboy how did creating a custom drone go? I'm looking into building one myself for simulating the jetson Tx2 board I have on board a larger drone than the standard drone in airsim.

Oskiboy, 2018-12-21:
@zungam is the one who fixed this!

AloshkaD, 2018-12-21:
@Oskiboy @zungam Thanks guys!

Ischio, 2019-11-13:
Did someone managed to do it?
I managed to modify masses, lengths and thust/toque parameters according to a real drone but the drone can't fly properly or, with other data, it doesn't take off. 
I think the problem is my drone has a mass of 0.087 (Weight I suppose, but in the SimpleFlightQuadXParameters is written mass ) while the standard one is 1 and it's written it can be modified from 0.8 to 1.6 .
Does it means I can't simulate my drone??
Thank you

chaitanyantr, 2020-04-23:
@Ischio By any chance you were able to fly the drone by changing parameters.Can you please share your experience on that.

Thankyou,
Chaitanya krishna

EdwinMeriaux, 2020-11-16:
@sytelus How would be create 3D mesh and Pawn BP?