microsoft/AirSim #840: Possible to add Cameras to the Drone?

olivia-skydio, 2018-02-22:
Is it possible to add a camera to the drone? What files would you have to change besides FlyingPawn.cpp? 
Thanks!

zergler, 2018-02-23:
You will have to modify the BP_FlyingPawn blueprint with the added camera and parameters, and then inside [FlyingPawn.cpp](https://github.com/Microsoft/AirSim/blob/e169e65755f4254c29b67617f50502a262e64e63/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.cpp#L46), add your reference to it, making sure to add it to the [camera list](https://github.com/Microsoft/AirSim/blob/58042d173096ac865c04daf2d9113004a7a23089/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.cpp#L19).

olivia-skydio, 2018-02-23:
@zergler thanks for the reply back! So I'm currently doing that right now but when I try and do ImageRequest for that camera then my UnrealEngine crashes. Any ideas of things I could be missing?

zergler, 2018-02-23:
Hmm I'm not sure. You could launch UE4Editor inside gdb and then run `r /path/to/airsim/Unreal/Environments/Blocks/Blocks.uproject` to get a stack track. That might point out the problem.

DavidLSmyth, 2018-02-23:
I have done this previously, you'll also need to make changes in [FlyingPawn.h](https://github.com/Microsoft/AirSim/blob/58042d173096ac865c04daf2d9113004a7a23089/Unreal/Plugins/AirSim/Source/Multirotor/FlyingPawn.h), [VehicleWrapper.cpp](https://github.com/Microsoft/AirSim/blob/ce85db7008b68316c4685a3990be6c9a032c3e53/Unreal/Plugins/AirSim/Source/VehiclePawnWrapper.cpp). I followed the code starting from [multiRotorAPI](https://github.com/Microsoft/AirSim/blob/ce85db7008b68316c4685a3990be6c9a032c3e53/AirLib/include/vehicles/multirotor/api/MultirotorApi.hpp) simGetImage method. I can outline the specific changes that I made if that isn't enough

olivia-skydio, 2018-02-23:
@DavidLSmyth thanks for the response! I made changes in FlyingPawn.h to add my camera. What other changes in VehicleWrapper.cpp would I need to add? Thanks so much!

olivia-skydio, 2018-02-23:
Just got it working I think! Thanks for the help @zergler! @DavidLSmyth  I didn't end up changing anything in VehicleWrapper.cpp was still curious what you changed?

sytelus, 2018-02-27:
Just curious, why you need additional cameras. We already have 5 cameras on the drone...

nnayeer, 2018-02-27:
Hi @sytelus , not the OP but might have an answer to your question. Since my team and I would like to duplicate camera rig configurations from real vehicles (we have many) and test new ones, it helps to be able to specify the exact number, position and orientation of each camera, which I'm currently attempting to do similar to the methods discussed on this thread. Even though my use is for the car, I'd imagine there's a similar reason for wanting this option on a drone.

sytelus, 2018-03-02:
For now you might be able to just change values here: https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp#L123

In long run we want to be able to specify this through json settings.

closed 2018-04-12 by sytelus

nalinraut, 2019-02-05:
@DavidLSmyth, Can you go over the changes or attach files to the comment so I can personally go through the changes? I need multiple stereo cameras on the multirotor. Thank you.

nalinraut, 2019-02-05:
@DavidLSmyth If I am not wrong, I can use the left and right camera as a stereo pair using the simGetImages API. Thank you!

nalinraut, 2019-02-11:
@olivia-skydio Did you only make changes to FlyingPawn.h or are there more changes to other scripts? Thank you!

madratman, 2019-02-12:
hey @nalinraut, see my comment on #1770 here https://github.com/Microsoft/AirSim/issues/1770#issuecomment-462949223