microsoft/AirSim #841: Is it possible to control the airsim with Arduino?

gitstar12, 2018-02-23:
I wanted to know how to control the airsim with the help of Arduino and a gyrosensor

lovettchris, 2018-02-27:
If Arduino can run python (micropython perhaps) then you could run the PythonClient scripts from an Arduino computer.  I know for sure you can do this from a RaspberryPi.  The only dependency is the msgpack-rpc-python.  So you may need to build that package if it isn't available for your specific board.

sytelus, 2018-02-27:
BTW, you can also use C++ API ([see here](https://github.com/Microsoft/AirSim/blob/master/HelloCar/main.cpp)). I believe C++ interface for msgpack-rpc is compilable for any platform supporting C++11 compilers.

gitstar12, 2018-03-08:
I am new to Airsim, so could you Kindly tell me more about this...

jonyMarino, 2020-09-18:
Closed due to inactivity. Feel free to reopen to ask again. Doc update pending.

closed 2020-09-18 by jonyMarino