microsoft/AirSim #846: Error while collecting images in hello_car.py

JedyYu, 2018-02-26:
I'm using /PythonClient/hello_car.py to run test on AirSimNH.exe and generate images. 

The car moves. However, it is stuck in errors in function simGetImages():

ERROR:tornado.application:Uncaught exception, closing connection.
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\site-packages\tornado\iostream.py", line 562, in wrapper
    return callback(*args)
  File "C:\Program Files\Python36\lib\site-packages\tornado\stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\msgpackrpc\transport\tcp.py", line 24, in on_read
    self.on_message(message)
  File "C:\Program Files\Python36\lib\site-packages\msgpackrpc\transport\tcp.py", line 29, in on_message
    raise RPCError("Invalid MessagePack-RPC protocol: message = {0}".format(message))
msgpackrpc.error.RPCError: Invalid MessagePack-RPC protocol: message = {'image_data_uint8': b'\x89PNG\r\n\x1a\n\x00\x00\x00\rIHDR\x00\x00\x01\x00\x00\x00\x00\x90\x08\x06\x00\x00\x00\xe7c\xb5\x91\x00\x00\x072IDATx\x01\xed\xdd\x......
......
\x82', 'image_data_float': [0.0], 'camera_position': {'x_val': 20.602811813354492, 'y_val': 6.341492176055908, 'z_val': -1.9228893518447876}, 'camera_orientation': {'w_val': 0.6760340332984924, 'x_val': -0.007432194426655769, 'y_val': 0.026305830106139183, 'z_val': 0.7363632321357727}, 'time_stamp': 464769951407030, 'message': '', 'pixels_as_float': False, 'compress': True, 'width': 256, 'height': 144, 'image_type': 3}
ERROR:tornado.application:Exception in callback functools.partial(<function wrap.<locals>.null_wrapper at 0x000001B347336510>)

I'm using python 3.6 and I'm sure numpy and msgpack-rpc-python is installed following the steps in https://github.com/Microsoft/AirSim/blob/master/docs/python.md.

What should I do? Thanks.

JedyYu, 2018-02-26:
BTW, I find if I remove everything except

from AirSimClient import *
client = CarClient()

before simGetImages(), it works. But if I add anything, it fails.

sytelus, 2018-02-27:
This is strange. I just tested this on latest version and seems to be working fine. I'm using Python 3.5 on Windows 10. Not sure if that is the difference... I would also suggest run from the latest source if you can in Blocks environment.

closed 2018-04-12 by sytelus