microsoft/AirSim #847: build.cmd issue

gitstar12, 2018-02-26:
[image]

lovettchris, 2018-02-27:
Fixed

gitstar12, 2018-02-27:
no its not fixed how to do it??

sytelus, 2018-02-27:
Looks like @FireTime fixed this with this PR: https://github.com/Microsoft/AirSim/pull/845

@lovettchris - is there a different fix?

FireTime, 2018-02-27:
Correct, I submitted that fix for this exact error.

gitstar12, 2018-02-28:
[image]

sytelus, 2018-03-02:
Please do not post screenshots. Take time to look at error messages and see if you can debug this by yourself. We can't reproduce this.

closed 2018-03-06 by gitstar12