microsoft/AirSim #851: Release v1.1.8 broke Python API for MultirotorClient timestampNow() method

wagnboss, 2018-02-27:
The following simple Python file runs without error when using release v1.1.7 and earlier environments:

from AirSimClient import *
import time

client = MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)
client.takeoff()

# how well does simulator time track computer time?
timestamp = client.timestampNow()
print(timestamp)

time.sleep(5)   #argument is # seconds

timestamp = client.timestampNow()
print(timestamp)

# quit cleanly
client.reset()
client.enableApiControl(False)
---------------------------------------------
Here is resulting output:

runfile('C:/Users/LocalUser/Documents/AirSim/time.py', wdir='C:/Users/LocalUser/Documents/AirSim')
Reloaded modules: AirSimClient
Waiting for connection: 
168237001000
173235001000
--------------------------------------------
However it fails when run with release v1.1.8 environments:

runfile('C:/Users/LocalUser/Documents/AirSim/time.py', wdir='C:/Users/LocalUser/Documents/AirSim')
Reloaded modules: AirSimClient
Waiting for connection: 
Traceback (most recent call last):

  File "<ipython-input-3-bfd3bed0bc36>", line 1, in <module>
    runfile('C:/Users/LocalUser/Documents/AirSim/time.py', wdir='C:/Users/LocalUser/Documents/AirSim')

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 710, in runfile
    execfile(filename, namespace)

  File "C:\ProgramData\Anaconda3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 101, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "C:/Users/LocalUser/Documents/AirSim/time.py", line 18, in <module>
    timestamp = client.timestampNow()

  File "C:\Users\LocalUser\Documents\AirSim\AirSimClient.py", line 494, in timestampNow
    return self.client.call('timestampNow')

  File "C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()

  File "C:\ProgramData\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)

RPCError: rpclib: server could not find function 'timestampNow' with argument count 0.

sytelus, 2018-03-01:
You are correct this API was removed because it is confusing. There is significant latency in server response so its value will never be correct. It looks like we didn't removed from AirSimClient. Just curious, are you using this API and why? If there is a reason we can consider adding it back again.

wagnboss, 2018-03-01:
No need for the API as timestamp is available in:
 ImageResponse class returned by AirSimClientBase class method simGetImages(requests)
 CarState class returned by CarClient class method getCarState() 
 MultirotorState class returned by MultirotorClient class method getMultirotorState()

closed 2018-03-01 by wagnboss