microsoft/AirSim #852: How to solve the build error of Airsim on win10

kdzy, 2018-03-01:
I followed the steps of tutorials in [https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md] to build AirSim on windows, but when I run the build.cmd, the errors appeared as below:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

 Is the reason I lack the installation of some library? If yes, what the library is? If not ,could anyone give me some suggestions?Thanks a lot!

sytelus, 2018-03-01:
Please confirm if you have VC++ included in Visual Studio 2015 Update 3. It looks like your VC++ compiler is of older version.

kdzy, 2018-03-01:
I have installed Visual Studio 2015 Update 3, and included  VC++. VC ++ version and VS2015 version is the corresponding.
[image]

kdzy, 2018-03-08:
I have solved the problem. The reason is that the VS 2015 was not installed properly.When I uninstalled the VS  that was installed before completely, and installed VS 2015 correctly, especially the VC++ complier,the error disappeared.

closed 2018-03-15 by sytelus