microsoft/AirSim #853: Create ROIs on images automatically based on a target's and camera's poses

HarrisSym, 2018-03-01:
I am trying to create ROIs (Regions Of Interest) on images automatically based on target's and camera's  poses using SimGetObjectPose in an API. The basic idea is to construct three matrices: camera_matrix, perspective_matrix and viewport_matrix. Then multiply the initial world coordinates of the target's bounding box with these matrices, in order to get the screen coordinates of the bounding box and finally create the ROI based on these coordinates.
Is there any easier way to have access to the screen coordinates on an object than above?

sytelus, 2018-03-02:
SimGetObjectPose  currently returns only NED coordinates. However its easy to extend APIs to get screen coordinates of object. To do that you can use GetActorPosition() followed by ProjectWorldLocationToScreen(). Also, see [this thread](https://answers.unrealengine.com/questions/41980/how-do-i-go-from-world-space-to-screen-space.html).

phildue, 2018-04-03:
Hi thanks for the suggested solution! I was able to extend the api and retrieve the projection ([fork](https://github.com/phildue/AirSim.git). Now I would like to draw the retrieved 2d coordinates in an image I get with getImage(). However, the coordinates seem to be off. How can I transform the coordinates to the reference frame of the image of one of the cameras (Front camera in my case).?

phildue, 2018-04-04:
So from what I understand ProjectWorldLocationToScreen() only works for the users viewport not for the images of the camera. If I want to project it on the camera plane I need the way @HarrisSym suggested. However, I tried multiple ways now to get the correct projection matrix following #269 either using the formula or estimating it from the chessboard. In all cases the projection is not correct. I manually projected some points on an empty image to verify my calculations and they seem fine. However, the view of the camera in unreal seems to be rendered differently. Any idea what I could be missing?

sytelus, 2018-10-17:
Hi there, it would be great to get PR if you have made any progress on this work so its useful to others as well. If this is still an open problem, please file a new issue.

closed 2018-10-17 by sytelus

phildue, 2018-10-17:
i have this running in an external client and I still need to specify the 3D coordinates of the object I am interested in, as well as the camera parameters. I guess these things are also available within Unreal/Airsim which would make it more usable. For a pull request I would need some help how to access and where to integrate.