microsoft/AirSim #856: rotateToYaw(yaw) returns at yaw but drone rotates more before stopping

wagnboss, 2018-03-01:
I am using the Python API.
MultirotorClient class method rotateToYaw(yaw, max_wait_seconds, margin) always over shoots the target yaw.  If large enough max_wait_seconds argument is supplied, method return when yaw is within margin of target yaw, but drone continues to rotate for a while. For example, see following pyton code:
from AirSimClient import *
import time
import math

factor = 180./math.pi

def print_info():
    pitch, roll, yaw = client.getPitchRollYaw()
    print("pitch %5.1f, roll %5.1f, yaw %5.1f" % (pitch*factor, roll*factor, yaw*factor))
    print()
    
client = MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

delay = 5
max_wait_seconds = delay

client.takeoff()
time.sleep(delay)
print_info()

print("rotateToYaw (90)")
client.rotateToYaw(90., max_wait_seconds, 1)
print_info()    # it reached 90
time.sleep(delay)
print_info()    # it continued rotating

-----------------------------
which produces the following output:
Waiting for connection: 
after takoff
pitch   0.0, roll   0.0, yaw   0.0

rotateToYaw (90)
pitch   0.0, roll   0.0, yaw  90.5

pitch   0.0, roll   0.0, yaw 114.4

-------------------------------
Is this is normal/expected/desired behavior? 
If so, how do I get the drone to STOP rotating when it reaches target yaw?  
I've tried adding the following line immediately after the call to rotateToYaw, but it does not work.

client.rotateByYawRate(0.,1)    #try to stop rotation

sytelus, 2018-03-02:
You might want to play around to turn PID controller. Overshoots might happen because p is too high for example. There is plenty of literature on web on PID tuning...

PID parameters: 
Play with rate and level PID parameters: https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/Params.hpp#L51

Code: https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/firmwares/simple_flight/firmware/AngleLevelController.hpp

mwanggh, 2019-10-31:
I found this command to work when the drone is hovering. If you control the head direction in move, you may use the drivetrain parameter. 

client.hoverAsync().join()
client.rotateToYawAsync().join

Yuchen-sky, 2020-03-18:
> I found this command to work when the drone is hovering. If you control the head direction in move, you may use the drivetrain parameter.
> 
> client.hoverAsync().join()
> client.rotateToYawAsync().join

i do as you said, but it is not work two airsim 1.2

cloudlakecho, 2020-07-02:
@wangnboss, Would you share your AirSim version? 

I don't have those Classes in my Python API. When I try to use `rotateToYaw` function, I receive this error:
```
client.rotateToYaw(math.radians(orientation[2]), max_wait_seconds, margin[2])
AttributeError: 'MultirotorClient' object has no attribute 'rotateToYaw'
```
It looks like that I don't have `getPitchRollYaw()` neither....

jonyMarino, 2020-11-18:
Solved by PR2516

closed 2020-11-18 by jonyMarino