microsoft/AirSim #858: error: "call is only supported for simulation."

melton1968, 2018-03-03:
I compiled AirSim on Mac/OSX with minor modifications to the setup.sh script and everything seems to compile correctly, but when I run the example DroneServer:

$ ./DroneServer 1
You are running in simulation mode.
find: /sys/bus/usb/devices/usb*/: No such file or directory
Server connected to MavLink endpoint at 127.0.0.1:14560
Hit Ctrl+C to terminate.

The corresponding HelloDrone command fails with the following:

$ ./HelloDrone
Waiting for connection - X
Connected!
Press Enter to get FPV image

Exception raised by the API, something went wrong.
rpclib: function 'simGetImages' (called with 1 arg(s)) threw an exception. The exception contained \
this information: getImageCapture() call is only supported for simulation.

Some pointers in a direction to investigate would be helpful.

sytelus, 2018-03-04:
Could you please share setup.sh changes?

Above you are running DroneServer which is supposed to be running on real vehicle. The "simulation" it supports is more like "test" mode and not real simulation. I agree that we should change verbiage here so there is no confusion. The image APIs requires Unreal engine simulation. Just curious what you are trying to do and want to accomplish...

melton1968, 2018-03-04:
You can find my changes to setup.sh and build.sh here: git@github.com:melton1968/AirSim.git

I adapted the build to my environment in the simplest way possible. I'd be happy to help you develop a more robust OSX build process if that would be useful. We would probably want to create a HomeBrew variant and a MacPorts variant. I think it would be pretty straight forward.

I'd like to run the vehicle simulation, but I obviously missed the basics on how everything wires together.  I have UE4 installed and working, but I am not sure how to start the simulation sever. A couple of pointers in the right direction would be helpful.

sytelus, 2018-03-06:
The better/robust OSX build would be awesome!

For car or drone simulation, you don't need to run DroneServer. I would suggest start with Blocks environment (follow [the steps for Linux](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md)). You basically just run Unreal project with AirSim plugin. The AirSim plugin will start the API server and then you can use [Python client](https://github.com/Microsoft/AirSim/blob/master/docs/python.md) to control the vehicle.

sytelus, 2018-04-12:
Please let us know any updates on sharing your setup.sh changes.

jonyMarino, 2020-08-12:
@melton1968 did you tried your modifications with an UE enviroment? if you didn't find anything I would close this issue.

jonyMarino, 2020-09-02:
Closed due to age and incompleteness.
Please feel free to open a new issue (refering this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-09-02 by jonyMarino