microsoft/AirSim #859: Unable to run 'make' when installing Unreal Engine

spidershu, 2018-03-03:
Hello
I am trying to run the `make` in for the Unreal Engine, however, I am not able to run it. 

When I run `make` by itself, I get the error

> bash "/home/nickshu/UnrealEngine/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClient Linux Development  
/usr/lib/mono/4.5/Microsoft.Common.targets: error : Access to the path "/home/nickshu/UnrealEngine/Engine/Source/Programs/DotNETCommon/DotNETUtilities/obj/Development/DotNETUtilities.csproj.FilesWrittenAbsolute.txt" is denied.  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, Boolean anonymous, FileOptions options) <0x7f52b64e22c0 + 0x005ed> in <filename unknown>:0 
  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, System.String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost) <0x7f52b64e2210 + 0x0004d> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream:.ctor (string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions,string,bool,bool,bool)
  at System.IO.StreamWriter.CreateFile (System.String path, Boolean append, Boolean checkHost) <0x7f52b6354b80 + 0x00095> in <filename unknown>:0 
  at System.IO.StreamWriter..ctor (System.String path, Boolean append, System.Text.Encoding encoding, Int32 bufferSize, Boolean checkHost) <0x7f52b6354840 + 0x00073> in <filename unknown>:0 
  at System.IO.StreamWriter..ctor (System.String path, Boolean append, System.Text.Encoding encoding, Int32 bufferSize) <0x7f52b6354800 + 0x00035> in <filename unknown>:0 
  at System.IO.StreamWriter..ctor (System.String path, Boolean append) <0x7f52b6354780 + 0x00032> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.StreamWriter:.ctor (string,bool)
  at Microsoft.Build.Tasks.WriteLinesToFile.Execute () <0x41449fe0 + 0x00103> in <filename unknown>:0 
/usr/lib/mono/4.5/Microsoft.Common.targets: error : Access to the path "/home/nickshu/UnrealEngine/Engine/Source/Programs/UnrealBuildTool/obj/Development/UnrealBuildTool.csproj.FilesWrittenAbsolute.txt" is denied.  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, Boolean anonymous, FileOptions options) <0x7f52b64e22c0 + 0x005ed> in <filename unknown>:0 
  at System.IO.FileStream..ctor (System.String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, System.String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost) <0x7f52b64e2210 + 0x0004d> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream:.ctor (string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions,string,bool,bool,bool)
  at System.IO.StreamWriter.CreateFile (System.String path, Boolean append, Boolean checkHost) <0x7f52b6354b80 + 0x00095> in <filename unknown>:0 
  at System.IO.StreamWriter..ctor (System.String path, Boolean append, System.Text.Encoding encoding, Int32 bufferSize, Boolean checkHost) <0x7f52b6354840 + 0x00073> in <filename unknown>:0 
  at System.IO.StreamWriter..ctor (System.String path, Boolean append, System.Text.Encoding encoding, Int32 bufferSize) <0x7f52b6354800 + 0x00035> in <filename unknown>:0 
  at System.IO.StreamWriter..ctor (System.String path, Boolean append) <0x7f52b6354780 + 0x00032> in <filename unknown>:0 
  at (wrapper remoting-invoke-with-check) System.IO.StreamWriter:.ctor (string,bool)
  at Microsoft.Build.Tasks.WriteLinesToFile.Execute () <0x41449fe0 + 0x00103> in <filename unknown>:0 
Failed to build to build tool (UnrealBuildTool)
Makefile:258: recipe for target 'CrashReportClient' failed
make: *** [CrashReportClient] Error 1

However, when I try to run `sudo make`, I get the following error
>bash "/home/nickshu/UnrealEngine/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClient Linux Shipping  
Building CrashReportClient...
Creating makefile for CrashReportClient (no existing makefile)
Performing full C++ include scan (no include cache file)
Using clang version '3.8.0' (string), 3 (major), 8 (minor), 0 (patch)
Building UnrealHeaderTool...
Target is up to date
Total build time: 0.15 seconds (NoActionsToExecute executor: 0.00 seconds)
Parsing headers for CrashReportClient
  Running UnrealHeaderTool CrashReportClient "/home/nickshu/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/CrashReportClient/Shipping/CrashReportClient.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -NoEnginePlugins
Refusing to run with the root privileges.
Error: UnrealHeaderTool failed for target 'CrashReportClient' (platform: Linux, module info: /home/nickshu/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/CrashReportClient/Shipping/CrashReportClient.uhtmanifest, exit code: Canceled (1)).
Makefile:252: recipe for target 'CrashReportClient-Linux-Shipping' failed
make: *** [CrashReportClient-Linux-Shipping] Error 1

I am not sure of what am I supposed to do. Can someone help me?
Thank you a million!

sytelus, 2018-03-04:
Please post pure Unreal related issues at http://answers.unrealengine.com/

mchen0037, 2018-03-19:
https://github.com/Microsoft/AirSim/issues/686 worked for me.

closed 2018-04-12 by sytelus

flegends, 2018-06-10:
@nicolasshu @sytelus 
 I have the same Issue @nicolasshu  ...

shuhelshaikh2, 2020-02-26:
sudo chown -R $USER <UE directory>

Guacamole-1, 2020-06-22:
https://forums.unrealengine.com/development-discussion/engine-source-github/91044-error-compiling-ue4-on-fedora-24
this will help you out it helped me :)

Rick-Weng, 2021-12-30:
similar wrong.
(base) hp@hp-pc:~/WXR/UE4/UnrealEngine-4.25$ make UE4Editor UE4Game UnrealPak CrashReportClient ShaderCompileWorker UnrealLightmass
bash "/home/hp/WXR/UE4/UnrealEngine-4.25/Engine/Build/BatchFiles/Linux/Build.sh" UE4Editor Linux Development  
Fixing inconsistent case in filenames.
Setting up Mono
Building UE4Editor...
Using 'git status' to determine working set for adaptive non-unity build (/home/hp/WXR/UE4/UnrealEngine-4.25).
Creating makefile for UE4Editor (no existing makefile)

Unhandled Exception:
System.ArgumentException: Path fragment '"CODE/\350\267\221\351\201\223\346\225\260\346\215\256\351\233\206/"' contains invalid directory separators.
  at Tools.DotNETCommon.FileSystemReference.CombineStrings (Tools.DotNETCommon.DirectoryReference BaseDirectory, System.String[] Fragments) [0x000f2] in <9ba951000a704e749df79fc161508246>:0 
  at Tools.DotNETCommon.FileReference.Combine (Tools.DotNETCommon.DirectoryReference BaseDirectory, System.String[] Fragments) [0x00000] in <9ba951000a704e749df79fc161508246>:0 
  at UnrealBuildTool.GitSourceFileWorkingSet.AddPath (System.String Path) [0x0002e] in <87819062dc15432b9a47ca49aa4b45d0>:0 
  at UnrealBuildTool.GitSourceFileWorkingSet.OutputDataReceived (System.Object Sender, System.Diagnostics.DataReceivedEventArgs Args) [0x000a2] in <87819062dc15432b9a47ca49aa4b45d0>:0 
  at System.Diagnostics.Process.OutputReadNotifyUser (System.String data) [0x00042] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.Diagnostics.AsyncStreamReader.FlushMessageQueue () [0x00046] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.Diagnostics.AsyncStreamReader.GetLinesFromStringBuilder () [0x00124] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.Diagnostics.AsyncStreamReader.ReadBuffer (System.IAsyncResult ar) [0x00126] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.IO.Stream+ReadWriteTask.InvokeAsyncCallback (System.Object completedTask) [0x00007] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.IO.Stream+ReadWriteTask.System.Threading.Tasks.ITaskCompletionAction.Invoke (System.Threading.Tasks.Task completingTask) [0x00043] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.FinishContinuations () [0x0007a] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.FinishStageThree () [0x0003c] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.FinishStageTwo () [0x000a1] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.Finish (System.Boolean bUserDelegateExecuted) [0x0002e] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.ExecuteWithThreadLocal (System.Threading.Tasks.Task& currentTaskSlot) [0x0004b] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.ExecuteEntry (System.Boolean bPreventDoubleExecution) [0x0004a] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentException: Path fragment '"CODE/\350\267\221\351\201\223\346\225\260\346\215\256\351\233\206/"' contains invalid directory separators.
  at Tools.DotNETCommon.FileSystemReference.CombineStrings (Tools.DotNETCommon.DirectoryReference BaseDirectory, System.String[] Fragments) [0x000f2] in <9ba951000a704e749df79fc161508246>:0 
  at Tools.DotNETCommon.FileReference.Combine (Tools.DotNETCommon.DirectoryReference BaseDirectory, System.String[] Fragments) [0x00000] in <9ba951000a704e749df79fc161508246>:0 
  at UnrealBuildTool.GitSourceFileWorkingSet.AddPath (System.String Path) [0x0002e] in <87819062dc15432b9a47ca49aa4b45d0>:0 
  at UnrealBuildTool.GitSourceFileWorkingSet.OutputDataReceived (System.Object Sender, System.Diagnostics.DataReceivedEventArgs Args) [0x000a2] in <87819062dc15432b9a47ca49aa4b45d0>:0 
  at System.Diagnostics.Process.OutputReadNotifyUser (System.String data) [0x00042] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.Diagnostics.AsyncStreamReader.FlushMessageQueue () [0x00046] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.Diagnostics.AsyncStreamReader.GetLinesFromStringBuilder () [0x00124] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.Diagnostics.AsyncStreamReader.ReadBuffer (System.IAsyncResult ar) [0x00126] in <23340a11bb41423aa895298bf881ed68>:0 
  at System.IO.Stream+ReadWriteTask.InvokeAsyncCallback (System.Object completedTask) [0x00007] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <98fac219bd4e453693d76fda7bd96ab0>:0 
[... cut ...]
  at System.Threading.Tasks.Task.ExecuteEntry (System.Boolean bPreventDoubleExecution) [0x0004a] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.Tasks.Task.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <98fac219bd4e453693d76fda7bd96ab0>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
make: *** [Makefile:928：UE4Editor] 错误 255