microsoft/AirSim #870: Can't reverse and a question

taszilvo, 2018-03-08:
Hi.
I can't go to reverse with my car using XBox360 controller. Is there any way to customize the controller button binding settings? Or how to reverse? :D
Thanks!

++: Do you see any possibility to implement car indexes for better simulation and / or give more telemetry data (like rpm, etc.)?

tiamotogether, 2018-03-09:
I am just interesting in multirotors, so I don't pay attention to cars.

mitchellspryn, 2018-03-14:
I've had luck with [x360ce](https://www.x360ce.com/) for customizing controls. Run it at the same time you're running AirSim.

sytelus, 2018-03-14:
B button switches to reverse gear, A button switches to forward gear.

See: https://github.com/Microsoft/AirSim/blob/fcb01f5c2eda5bb9f7b9afffa539f6119382c741/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp#L433

closed 2018-04-12 by sytelus