microsoft/AirSim #873: Version Issue with Tornado

named-user, 2018-03-09:
The latest version of Tornado (5.0) deprecates the ioloop, which is used with msgpack, resulting in an error when hello_drone.py is run.
Reverting to Tornado 4.5 solves the issue.
The issue seems related to the pip install of msgpack-rpc and its tornado dependencies.

Platform:
Linux
Offline
Manually installing dependencies

Error:
```
Traceback (most recent call last):
  File "./hello_drone.py", line 13, in <module>
    client = MultirotorClient()
  File "~/AirSim/PythonClient/AirSimClient.py", line 482, in __init__
    super(MultirotorClient, self).__init__(ip, 41451)
  File "~/AirSim/PythonClient/AirSimClient.py", line 196, in __init__
    self.client = msgpackrpc.Client(msgpackrpc.Address(ip, port), timeout = 3600, pack_encoding = 'utf-8', unpack_encoding = 'utf-8')
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/client.py", line 15, in __init__
    loop.attach_periodic_callback(self.step_timeout, 1000) # each 1s
  File "/usr/local/lib/python3.5/dist-packages/msgpackrpc/loop.py", line 38, in attach_periodic_callback
    self._periodic_callback = ioloop.PeriodicCallback(callback, callback_time, self._ioloop)
TypeError: __init__() takes 3 positional arguments but 4 were given
```

Lance804, 2018-03-11:
Thank you very much. I thought the tornado under 4.1 is needed, but it is conflict.

sytelus, 2018-03-14:
So what is the solution here? May be msgpack-rpc should force specific version?

sytelus, 2018-04-12:
Looks like this is fixed: https://github.com/msgpack-rpc/msgpack-rpc-python/issues/19

So if you update the package then it should work.

closed 2018-04-13 by sytelus