microsoft/AirSim #874: Packing Airsim in linux

geevargs, 2018-03-09:
Has anyone packages an Airsim game for linux sucessfully?
I am not able to package it.
It works when I remove Airsim but with Airsim I keep getting errors.
UATHelper: Packaging (Linux): UnrealBuildTool: clang: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Packaging (Linux): UnrealBuildTool: ERROR: UBT ERROR: Failed to produce item: /home/sujith/MyProject2/Binaries/Linux/MyProject2-Linux-DebugGame
UATHelper: Packaging (Linux): UnrealBuildTool: Total build time: 783.01 seconds (Local executor: 0.00 seconds)
UATHelper: Packaging (Linux): CommandUtils.Run: Run: Took 783.371014s to run mono, ExitCode=5
UATHelper: Packaging (Linux): Program.Main: ERROR: AutomationTool terminated with exception: AutomationTool.CommandUtils+CommandFailedException: Command failed (Result:5): /home/sujith/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe MyProject2 Linux DebugGame -Project=/home/sujith/MyProject2/MyProject2.uproject  /home/sujith/MyProject2/MyProject2.upr
oject -NoUBTMakefiles  -remoteini="/home/sujith/MyProject2" -skipdeploy -noxge -NoHotReload -ignorejunk. See logfile for details: 'UnrealBuildTool-2018.03.08-18.19.57.txt' 
UATHelper: Packaging (Linux):   at AutomationTool.CommandUtils.RunAndLog (System.String App, System.String CommandLine, System.String Logfile, Int32 MaxSuccessCode, System.String Input, ERunOptions Options, System.Collections.Generic.Dictionary`2 EnvVars, AutomationTool.SpewFilterCallbackType SpewFilterCallback) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.CommandUtils.RunAndLog (AutomationTool.CommandEnvironment Env, System.String App, System.String CommandLine, System.String LogName, Int32 MaxSuccessCode, System.String Input, ERunOptions Options, System.Collections.Generic.Dictionary`2 EnvVars, AutomationTool.SpewFilterCallbackType SpewFilterCallback) [0x0000
0] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.CommandUtils.RunUBT (AutomationTool.CommandEnvironment Env, System.String UBTExecutable, System.String CommandLine, System.String LogName, System.Collections.Generic.Dictionary`2 EnvVars) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.CommandUtils.RunUBT (AutomationTool.CommandEnvironment Env, System.String UBTExecutable, UnrealBuildTool.FileReference Project, System.String Target, System.String Platform, System.String Config, System.String AdditionalArgs, System.String LogName, System.Collections.Generic.Dictionary`2 EnvVars) [0x00000] in
 <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.UE4Build.BuildWithUBT (System.String TargetName, UnrealTargetPlatform TargetPlatform, System.String Config, UnrealBuildTool.FileReference UprojectPath, Boolean ForceMonolithic, Boolean ForceNonUnity, Boolean ForceDebugInfo, Boolean ForceFlushMac, Boolean DisableXGE, System.String InAddArgs, Boolean ForceUnity
, System.Collections.Generic.Dictionary`2 EnvVars) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.UE4Build.Build (AutomationTool.BuildAgenda Agenda, Nullable`1 InDeleteBuildProducts, Boolean InUpdateVersionFiles, Boolean InForceNoXGE, Boolean InUseParallelExecutor, Boolean InForceNonUnity, Boolean InForceUnity, Boolean InShowProgress, System.Collections.Generic.Dictionary`2 PlatformEnvVars, Nullable`1 InC
hangelistNumberOverride, System.Collections.Generic.Dictionary`2 InTargetToManifest) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at Project.Build (AutomationTool.BuildCommand Command, AutomationTool.ProjectParams Params, Int32 WorkingCL, ProjectBuildTargets TargetMask) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at BuildCookRun.DoBuildCookRun (AutomationTool.ProjectParams Params) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at BuildCookRun.ExecuteBuild () [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.BuildCommand.Execute () [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.Automation.Execute (System.Collections.Generic.List`1 CommandsToExecute, System.Collections.Generic.Dictionary`2 Commands) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.Automation.Process (System.String[] Arguments) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.Program.MainProc (System.Object Param) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.InternalUtils.RunSingleInstance (System.Func`2 Main, System.Object Param) [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux):   at AutomationTool.Program.Main () [0x00000] in <filename unknown>:0 
UATHelper: Packaging (Linux): Program.Main: AutomationTool exiting with ExitCode=5 (5)
UATHelper: Packaging (Linux): RunUAT ERROR: AutomationTool was unable to run successfully.
PackagingResults: Error: Unknown Error
LogLinux: Setting swap interval to 'Immediate'

M-Kasem, 2018-03-10:
somewhat similar issue here, here his the packaging output: 
[package.txt](https://github.com/Microsoft/AirSim/files/1799488/package.txt)

OS: ubuntu 16.04
Environment: Blocks
Unreal: 4.17

sytelus, 2018-03-14:
Please post the full output. You want to get the first error message.

mitchellspryn, 2018-03-14:
I have gotten cross compilation working.

There are a few problems. First, cross compilation doesn't work with binary builds of the unreal engine, so you need to build the engine from source.

Next, mavlinkcom doesn't have linux binaries available to download, so you'll need to build AirSim on your target linux platform and copy-paste LibMavLinkCom.a to the root directory of your cross compiler. I haven't figured out how to get the compiler to build LibMavLinkCom along with the other binaries. 

The final problem I ran into is that you need some custom compile flags. The easiest way to do this is to edit LinuxToolChain.cs, which contains the logic for building up the compiler commands. I had to add the following compiler flags:

 -lstdc++ -std=c++0x -lsupc++ -v

Note that you'll have to add these in both GetLinkArguements() and LinkFiles(). Note that building with the custom engine will take a lot longer than with the binaries.

sytelus, 2018-03-14:
@M-Kasem , @geevargs 
I believe this was fixed ~15 days ago: https://github.com/Microsoft/AirSim/commit/85ac643bd1e89f7481e5e2b04408b06efca553ac

Could you please confirm you are using latest code?

geevargs, 2018-03-15:
@sytelus I used the code below  
I opened the dir  /AirSim/external/rpclib/rpclib-2.2.1/lib/rpc/ gedit this_handler.cc
extern "C" int __cxa_thread_atexit(void (*func)(), void *obj,
                                   void *dso_symbol) {
  int __cxa_thread_atexit_impl(void (*)(), void *, void *);
  return __cxa_thread_atexit_impl(func, obj, dso_symbol);
}

sytelus, 2018-03-15:
@geevargs - do you mean you changed the code and fixed the issue? I'm not following you.

geevargs, 2018-03-15:
Yes...I changed the code and that fixed it...

sytelus, 2018-03-15:
Great! Could you please submit PR?

stanleycelestin1, 2018-07-05:
@geevargs  where in this _handler.cc are those lines you provided supposed to go?

stanleycelestin1, 2018-07-05:
Were there any questions required? @geevargs

stanleycelestin1, 2018-07-05:
Were there any additional changes required?

madratman, 2019-04-10:
you can use this script to package unreal projects in linux https://github.com/Microsoft/AirSim/blob/master/docs/docker_ubuntu.md#misc-packaging-unreal-environments-in-airsim_source-containers

closed 2019-04-10 by madratman