microsoft/AirSim #876: Accessing the APIs via C#

henrik94, 2018-03-12:
I've tried several methods for calling the API from a Visual C# project, but I just couldn't make it work properly. Could you provide a small example how should you start if you want to access the API via C#? Some more documentation on this would be useful.

sytelus, 2018-03-14:
You might want to look at https://github.com/yfakariya/msgpack-rpc-cli

xuanmung, 2018-04-02:
I've the same problem. Have you solved it, @henrik94 ?
Any body could provide a particular API for C# plz?
Thanks in advance~

totht91, 2018-04-02:
We have made a good progress and we are considering to make it public after some more stabilization.

xuanmung, 2018-04-03:
Great, @totht91. I am really looking forward to your contribution.

NextSim, 2018-04-05:
Excited about using this as well.

sytelus, 2018-04-12:
@NextSim had been looking in to this as well: https://github.com/Microsoft/AirSim/issues/946

Any updates?

NextSim, 2018-04-12:
I took a look at the quickstart guide 

https://github.com/yfakariya/msgpack-rpc-cli/wiki/Quick-start

but after installing the NuGet package for MsgPack.Cli, it doesn't contain a definition for RpcClientConfiguration. 

Maybe I'm missing something

NextSim, 2018-04-19:
I finally got communication working over the api with c#.  

I used the MessagePack serializer below

https://github.com/neuecc/MessagePack-CSharp

For the enable api  and set car control I created the classes and helper functions below.

```
using MessagePack;
namespace Simulator
{
    [MessagePackObject]
    public class carControlHelper
    {
        [Key(0)]
        public controlApiParams apiParams { get; set; }
    }

    [MessagePackObject]
    public class enableApiHelper
    {
        [Key(0)]
        public bool val { get; set; } 
    }

    [MessagePackObject]
    public class controlApiParams
    {
        [Key("throttle")]
        public float throttle { get; set; }

        [Key("steering")]
        public float steering { get; set; }

        [Key("brake")]
        public float brake { get; set; }

        [Key("handbrake")]
        public bool handbrake { get; set; }
        [Key("is_manual_gear")]
        public bool is_manual_gear { get; set; }
        [Key("manual_gear")]
        public int manual_gear { get; set; }
        [Key("gear_immediate")]
        public bool gear_immediate { get; set; }
    }

    public static class msgPackRpcHelper
    {
        private static int msgCounter = 0;

        public static byte[] enableApiHelper(int type, bool enable)
        {
            var helperObj = new object[] { type, msgCounter++, "enableApiControl", new enableApiHelper { val = enable } };
            return MessagePackSerializer.Serialize(helperObj);
        }

        public static byte[] setCarControlHelper(int type, float accel, float strAngle, float brake, bool handbrake, bool isManual, int gear, bool gearImmediate)
        {
            var helperObj = new object[] { type, msgCounter++, "setCarControls", new carControlHelper{apiParams = new controlApiParams {
                    throttle = accel,
                    steering = strAngle,
                    brake = brake,
                    handbrake = handbrake,
                    is_manual_gear = isManual,
                    manual_gear = gear,
                    gear_immediate = gearImmediate
                } } };
            return MessagePackSerializer.Serialize(helperObj);
        }
    }
}
```

sytelus, 2018-04-25:
Very nice! We look forward to your pull request :).

NextSim, 2018-04-26:
Working on it.  It's incomplete though, since classes have to be made for each api call and response, but there aren't any details on each api call so it is slow developing it.

brandtd, 2018-05-01:
@NextSim, I've been trying to do something similar, and I've made some progress calling the RPCs from C#. So far, though, I've been focused on the multirotor side of things. If you've got the time, you may want to take a look, specifically at the AirSimRpc and MsgPackRpc projects: https://github.com/brandtd/AirSimApp

zimmy87, 2021-03-12:
closed to age and lack of activity

closed 2021-03-12 by zimmy87

tank104, 2022-11-28:
@NextSim are you able to share the code that sends the message you have above?
I am trying to use MagicOnion (which uses MessagePack) but it keeps crashing AirSim whenever I make an API call.

tank104, 2022-12-09:
@brandtd @xuanmung  did you manage to get working? any code you can share?

tank104, 2022-12-11:
For anyone that sees this - I managed to get it working with the following code below.
I went down a bit of rabbit hole with more recent RPC libraries that just weren't compatible (MagicOnion and streamjsonrpc).

[Include https://github.com/yfakariya/msgpack-rpc-cli]

using MsgPack.Rpc.Client;
using MsgPack.Serialization;
using System.Net;

namespace TestConsoleApp
{
  internal class Program
  {
    static void Main(string[] args)
    {
      var configuration = new RpcClientConfiguration();
      var context = new SerializationContext();
      context.SerializationMethod = SerializationMethod.Map;
      using (dynamic proxy = new DynamicRpcProxy(new DnsEndPoint("127.0.0.1", 41451), configuration, context))
      {
        int msgCounter = 0;
        var result = proxy.isApiControlEnabled("");

        var imageResponse = proxy.simGetImages(new ImageRequest[]
        {
          new ImageRequest
          {
            camera_name = "0",
            image_type = 0,
            pixels_as_float = false,
            compress = false
          }
        }, "", false);
      }
    }
  }

  public class ImageRequest
  {
    public string camera_name { get; set; }
    public int image_type { get; set; }
    public bool pixels_as_float { get; set; }
    public bool compress { get; set; }
  }
}