microsoft/AirSim #880: write a function that will move the drone w.r.t its own referential?

Pierrolo, 2018-03-13:
Hello!
I am wondering the exact same thing. I'm having trouble understanding how you can have a control command that says "go right" (NOT go east) no matter where the front camera is pointing. Since the input of my model is the front cam then with the same input (say facing a wall) it should choose the same action (meaning choose the same value to a certain action-control function) that should have the same consequence : "go right".
I tried using this function (I'm using Pyhton API) :

`def run_a_pov(client , action = [0,1,0] , duration = 5 , drivetrain = DrivetrainType.ForwardOnly , yaw_mode = YawMode(is_rate = False, yaw_or_rate = 0.0)):`
    `q = client.getOrientation()`
    `my_quaternion = Quaternion(w_val=q.w_val,x_val=q.x_val,y_val= q.y_val,z_val=q.z_val)`
    `mvm = my_quaternion.rotate(action)`
    `client.moveByVelocity(mvm[0],mvm[1],mvm[2], duration = 5 , drivetrain =  DrivetrainType.ForwardOnly , yaw_mode = YawMode(is_rate = False, yaw_or_rate = 0.0) )`

And if you try running this function for: 

`run_a_pov(client , action = [0,1,0] , duration = 5 , drivetrain = DrivetrainType.ForwardOnly , yaw_mode = YawMode(is_rate = False, yaw_or_rate = 0.0))`
 you will notice that the first few calls make the drone go right (yipee) and then it start going left instead and sooner or later it will just start doing circles.

Sorry about the long post, here's how I would sum it up:

How can I write a function that will move the drone w.r.t its own referential? Meaning : 
if I say [x,y,z] it will apply x, y and z on the drone point of view; not on the wold referential . The idea is to train the drone in simulation and then do some transfer learning in order to spend less time training it in real life. And so any action specified should be with regard to what the drone is seeing.
Thank you very much for you help, it would be much appreciated

sytelus, 2018-03-13:
This is called "headless" mode and nowadays its implemented even in $50 drones. The way it works is simple. You want drone to go in direction of unit vector that is in world frame. You you convert that in to drone's body frame and use moveByVelocityZ command. [Code here does that](https://github.com/Microsoft/AirSim/blob/master/AirLib/src/vehicles/multirotor/controllers/DroneControllerBase.cpp#L458). If you create a sample, please do submit the PR :).

Pierrolo, 2018-03-14:
Hello,
Thanks for the answer (very very appriciated :) ).
What is the difference between moveByVelocityZ  and moveByVelocity? 
Attached what I have so far ([sample_airsim.zip](https://github.com/Microsoft/AirSim/files/1809963/sample_airsim.zip)), and that seems to be working : the same actions have the same consequences in body_frame. So I seem to finally have my action state space : `[+North , +East, +Down, Angle]`, where direction are in m/s and Angle is in degree/s. My next problem is figuring out which `sleepy_time ` to take in : 
`for i in range (100): `
   `actions = [0.5, 0.3, -0.2, 0]
    run_a_pov(client , actions = actions)
    time.sleep(sleepy_time)`

I am trying to do Reinforcement Learning on that.

Thanks :)

sytelus, 2018-03-14:
btw moveByVelocityZ  is also implemented for simple_flight :)

closed 2018-04-12 by sytelus

flyers, 2018-12-17:
@Pierrolo Your code is very helpful! Thanks!