microsoft/AirSim #883: I downloaded the lastest release source code,and what should I do than to ran AirSim.

jinlukang1, 2018-03-14:
I have run the install_run_all.sh,but after that I don't know what to do to open AirSim.

Can someone can tell me what should I do next?

mitchellspryn, 2018-03-14:
I think the .sh files are intended to be used with linux. AirSim is a plugin, so roughly what you'll need to do is run Build.cmd, which will place the plugin content into Unreal/Plugins. Then, copy that into your unreal project into the plugins folder, edit the .uproj file to enable the plugin, start the project, and set your game mode to "AirSimGameMode". These steps are detailed [here](https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md). 

Alternatively, you can download the [pre-built binaries](https://github.com/Microsoft/AirSim/releases) if you just want to experiment.

jinlukang1, 2018-03-16:
Thanks for your reply.
But when I run the Build.cmd on Windows10, the Cmd shows this.
```
iwr : 请求被中止: 未能创建 SSL/TLS 安全通道。
所在位置 行:1 字符: 5
+ & { iwr https://github.com/rpclib/rpclib/archive/v2.2.1.zip -OutFile  ...
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-WebRequest]，WebExce
    ption
    + FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand
```
Can you figure out what's wrong?

mitchellspryn, 2018-03-16:
[What version of powershell do you have installed](https://stackoverflow.com/questions/1825585/determine-installed-powershell-version)? If it's not 5.0, try [downloading the newest version](https://docs.microsoft.com/en-us/powershell/scripting/setup/installing-windows-powershell?view=powershell-6)

closed 2018-04-10 by jinlukang1