microsoft/AirSim #885: Slow rendering on a ROS topic

kshitijgoel007, 2018-03-14:
Hi!

Thanks for this great plugin. I am playing around with PythonClient and integrating image rendering (using `simGetImage`) to one of my ROS nodes using `rospy`.

However, I am observing very low FPS (< 10 FPS) even for a QVGA resolution image (both `Scene` and `DepthPerspective`). My config is `GTX 970` , `Core™ i7-4770 CPU @ 3.40GHz × 8` , `16GB RAM`. I am using the freely available LandscapeMoutains environment.

Is this bottlenecked by GPU usage? or am I doing something wrong? Should I use the C++ client instead?

sytelus, 2018-03-14:
Could you try record button and see the FPS you get? I suspect this might be delay in your pipeline (python to AirSim, for example). The latest version we released last week gets 30 FPS fairly easily in recording mode.

kshitijgoel007, 2018-03-26:
Closing this as I resolved the problem by slowing down my ROS clock speed.

closed 2018-03-26 by kshitijgoel007

pengjili, 2019-03-15:
@sytelus can I ask what hardwares you are using? the latest version I am using now is facing a problem of very slow recording, definitely much far away from 30 FPS. (Especially want to know what video card you use?)