microsoft/AirSim #886: Use fixed simulation step

zbenic, 2018-03-14:
Hello again.

I've asked the same thing here: https://github.com/Microsoft/AirSim/issues/808, but you can close that issue because I have managed to implement it.

My next step would be to implement the functionality that is mentioned in issue:
https://github.com/Microsoft/AirSim/issues/600

What I would like to do is to get the rotor speeds via the reinforcement learning, apply them by using movebyrotorspeed command, and get the state from the environment.
I would like to do that with the fixed timestep dt.

Can you give me a few pointers?

sytelus, 2018-03-14:
Could you please submit PR for https://github.com/Microsoft/AirSim/issues/808 ?

The simple_flight already works with fixed dt. I think what you mean is ability to step through via API. I think I have an idea. Right now AirSim supports different types of clocks which provides timing. All clocks should be derived from ClockBase class. We can create a new type of clock called ManualClock that simply doesn't do anything. You can just set whatever time you want in it and it returns you precisely same time you had set until you set the new value. Now you can have API called setManualClock which allows you to advance the clock.

zbenic, 2018-03-16:
Thanks for the quick responce.
I will submit PR during the day.

Just to check if the proposed solution would work for me, here is more detailed description of my idea.

What I want to achieve is to write the main() which will handle RL calls and which will communicate with the Blocks environment.
For each clock->step() in the main() I would get the actions (rotor RPMs) using RL.
Those speeds would be arguments to my moveByRotorSpeed command which duration would be equal to the clock step.
Blocks environment would receive command and the quadcopter would move according to the engine physics.
I would return the new state back to my main script.
The new state would be further used to check if it is terminal state, to get reward and later to get new actions if the state is not terminal.
I would like for quadcopter in Blocks to just wait for the new clock step and new moveByRotorSpeed, everything paused, so it can just continue when the next clock->step() occurs.
The simulation would run step by step.

Can that be achieved with your proposed solution?

Feel free to ask if something is unclear.

sytelus, 2018-03-23:
I think APIs should be just left alone and shouldn't have parameters for clock speed. The clock speed is controlled using separate API call. I'm planning to work on this shortly...

zbenic, 2018-03-23:
Can you elaborate that a bit?

Which API call is controlling the clock speed?
And can I update/get data from the Blocks environment incrementally, independently of the real time clock?