microsoft/AirSim #887: How to get the real distance

dehuachen2017, 2018-03-16:
Hey!
I want to know the distance from the drone to the view.
But I do not know how to get the real distance.
Thanks.

allen9273, 2018-03-16:
I am wondering, too.

sytelus, 2018-03-23:
What do you mean distance from "the view"? Can you elaborate on your goal?

allen9273, 2018-03-23:
I'd like to recognize humans in the environment. Upon this, flying to the front of the human and maintain a specific distance from the human. How can I approach this?

dehuachen2017, 2018-03-27:
for example, , as following depth map
[]([]([image]))

sytelus, 2018-04-12:
Please look at HelloDrone.py example that shows how to get depth map as floats 2D numpy array. Each cell value then is distance.

closed 2018-04-12 by sytelus

dehuachen2017, 2018-04-12:
Thank you for your reply.