microsoft/AirSim #888: Face camera downwards

guillem74, 2018-03-18:
Hey, I am trying to get aerial images from the drone, so I need to face the camera downwards.

I tried this setCameraOrientation method: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#camera-orientation-gimble

client.setCameraOrientation(0, AirSimClientBase.toQuaternion(0.261799, 0, 0)); #radians

But it throws this error to me:

client.setCameraOrientation(0, AirSimClientBase.toQuaternion(0.261799, 0, 0)); #radians

aburgm, 2018-03-19:
Camera index 3 is already downward facing, so you if you capture images from that one, you don't need to change any camera orientation at all.

guillem74, 2018-03-19:
Camera index 3 is giving me aerial images but I see the drone in them. I need to get images faced downwards FROM the drone.

guillem74, 2018-03-19:
@aburgm Thanks, I changed to camera number 2 and it is more or less what I was looking for.  For instance, calling: responses = client.simGetImages([ImageRequest(2, AirSimImageType.Scene)])

Do you know at which rate is the drone getting images? I maybe want to set it to just obtain images 1 per second, and therefore I can process them in the time between. I am trying to get aerial images for image classification using neural networks approach.

aburgm, 2018-03-20:
Strange, 3 works fine for me for downwards imagery. To obtain 1 image per second, just call the `simGetImages` function once a second.

closed 2018-03-23 by sytelus

deelakavasura, 2020-03-03:
When I use camera 3 without a gimbal if faces downwards. When the gimbal is on it faces forward. I want the downward facing camera with the gimbal effect. How to do this?