microsoft/AirSim #889: Integrating human players to AirSim

viniciusguigo, 2018-03-19:
Hello, everyone.

What would have to be done in order to integrate a human player (similar to the first-person view Unreal template) together with the quadrotor in some sort of multi-agent scenario? How to contribute to this effort?

Thanks!

sytelus, 2018-03-23:
There is a manual mode (press M key) which allows you to use arrow keys + AWSD keys to control view manually. Is that what you are thinking about.

viniciusguigo, 2018-03-24:
Hello, sytelus.

I believe I was not able to enter into Manual mode pressing M. Not even after changing to "ViewMode": "Manual" in the settings file. Am I missing anything?

Does it allow you to control the camera as it was gimballed (attached to the quadrotor) or does it allow to move freely around the map while the quadrotor operates normally with their onboard cameras?

Thanks!

sytelus, 2018-04-12:
Just tested it and M key works as expected. It controls the external camera (not the camera on drone).

closed 2018-04-12 by sytelus