microsoft/AirSim #893: libc++ / libstdc++ compatibility issues

mescarra, 2018-03-20:
Hello everyone,

I'm trying to compile a simple example, like the HelloDrone, but also adding any other dynamic library you could find in linux (built with gcc and libstdc++). I just cannot get it done.

I'm using Ubuntu 16.04 on an ARM64 architecture. I have installed AirSim's static libraries in /usr/local/lib.

So far, I've tried to use different combinations of g++, clang++-3.9, stdlib=libstdc++/libc++ and selective static/dynamic linking (-Wl,-Bstatic).

If I compile this way:
`g++ -I/opt/ros/kinetic/include  -I/usr/include/eigen3 -I/usr/local/include/airsim -I/usr/local/include/rpc -DRPCLIB_MSGPACK=clmdep_msgpack -isystem /usr/local/include/llvm -std=c++11 -DMSGPACK_PP_VARIADICS_MSVC=0 -fPIE -o Drone main.cpp -L/opt/ros/kinetic/lib -lrosbag_storage -lc++ -lc++abi -lpthread -lAirLib -lrpc -lMavLinkCom` (`lc++` and `lc++abi` are llvm compiled libraries.)

I get:

`undefined reference to `rosbag::Bag::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int)'`

which is reasonable because
`nm -C -D /opt/ros/kinetic/lib/librosbag_storage.so | grep rosbag::Bag::open`

outputs

`/opt/ros/kinetic/lib/librosbag_storage.so:0000000000023080 T rosbag::Bag::open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)`

On the other hand, if I remove libc++ link (so defaults to libstdc++) I get undefined errors in rpc library like so:
`../build_debug/output/lib/librpc.a(format.cc.o): In function 'void clmdep_fmt::(anonymous namespace)::check_sign<wchar_t>(wchar_t const*&, clmdep_fmt::internal::Arg const&)':
.../rpclib-2.2.1/dependencies/src/format.cc:306: undefined reference to 'std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'`

Any ideas?

Thanks in advance.

mescarra, 2018-03-20:
Well, I've made some progress.

I decided to rebuild AirSim with `g++` and `libstdc++`. I just had to modify the root `build.sh` script to select the compiler and a few lines of code calling `std::fabsf` which is not part of the std namespace. And that's all.

Now linking against g++ compiled libraries works great, and I was able to connect the UnrealEngine server compiled by default with my g++ MultirotorClient, and so far so good.

I'll let you know if I run into any issue of this kind ever again.

sytelus, 2018-03-23:
That is great! Would you be able to create a pull request with your changes so may be other users can also benefit?

sytelus, 2018-04-12:
PR was merged. Thanks!

closed 2018-04-12 by sytelus