microsoft/AirSim #895: update_from_git error

hyybuaa, 2018-03-21:
If you are looking for a new feature or bug fix, fire up your IDE, try playing around with source code and contribute back! 
What's better than filing issue? Filing a pull request :).

---- (Remove above before filing the issue) -----
Get-ItemProperty : 找不到路径“HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj”，因为该路径不存在。
所在位置 行:1 字符: 6
+ & { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile ...
+      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (HKEY_CLASSES_RO...hell\rungenproj:String) [Get-ItemProperty], ItemNotFo
   undException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.GetItemPropertyCommand

sytelus, 2018-03-23:
Please use regedit to see if following path exist in registry: `HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj`

Chunchaie, 2018-03-27:
Some issue happened to me. Any updates?

hyybuaa, 2018-03-28:
@Chunchaie  i didn't follow the official tutorial and given up the steap, i used the vs 2015_enterprise instead of the community,
this is a  feasible tutorial: http://www.cnblogs.com/frankfanxinqi/p/8119931.html

closed 2018-03-28 by hyybuaa

Chunchaie, 2018-03-29:
@hyybuaa - OK. Thank you.

woodenchair, 2024-05-25:
i just find a perfect solution for the issue: https://zhuanlan.zhihu.com/p/147660674