microsoft/AirSim #897: Steering wheel on Ubuntu

EdnilsonAraujo, 2018-03-21:
Hello. AirSim really is a great job. I intend to use it in my research with an autonomous car. I need to get a steering wheel + pedal assembly. Do you suggest any models that are compatible with AirSim + Ubuntu? I saw in the image AirSimCarManual.gif that you used a steering wheel. Could you tell me the model of the steering wheel you used? Tks.

sytelus, 2018-03-23:
We have used Logitch G920 and it works out of the box. However we have tested only under Windows but you might be able to work in Linux with little or no changes. You will need to following settings to enable steering wheel controls:

```
"PhysXCar" : {
    "RC": {
        "RemoteControlID": 0
    }
}
```

BrianN92, 2018-04-12:
Hi @sytelus, after adding those settings to "settings.json", the car starts to turn left and have a steering angle "-1.25" every time I run the simulator. Some bottons and the steering wheel work now, but the car is out of control.

EdnilsonAraujo, 2018-04-18:
Hi @sytelus, thanks for help me!

DrifterFun, 2018-06-14:
@BrianN92 I meet the same problem just as you man. You can do some change in file "CarPawn.cpp" as following:
```
//joystick_controls_.steering = joystick_state_.left_y * 1.25;
//joystick_controls_.throttle = (-joystick_state_.right_x + 1) / 2;
joystick_controls_.steering = joystick_state_.left_x * 1.25;
joystick_controls_.throttle = (joystick_state_.left_y + 1) / 2;
joystick_controls_.brake = -joystick_state_.right_z + 1;
```

BrianN92, 2018-07-18:
Thanks @DrifterFun ! I tried, but it didn't work...

wintersoom, 2019-07-09:
We have used Logitch G29 by the steps of https://microsoft.github.io/AirSim/docs/steering_wheel_installation/
it did not work,the speed is always 0,i can not move the car.
what can i do? @sytelus