microsoft/AirSim #899: AirSim plugins can't be enabled on Ubuntu

kdzy, 2018-03-22:
I have installed AirSim and UE4 successfully on Ubuntu 14.04, and the Built-in Blocks Environment can run normally;While I set my own environment, the following appeared:
[image]
If I choose yes, then the environment can be open without airsim enabled as following . When I button Enable, the UE4 restarted and the above phenomenon appeared again;
[image]
 If I choose no, the following appeared; and the project can't be  compiled,　and it prompts "try rebuilding from source manually", what should I do to solve the problem?
[image]

The version is as followed:
Ubuntu:14.04;
Unreal Engine:4.17;

settings.json:

             {
             "SettingsVersion": 1.0,
              "DefaultVehicleConfig": "SimpleFlight",
                "SimpleFlight": {
             "RemoteControlID": 0,
           "RC": {
                   "AllowAPIWhenDisconnected": true,
                   "AllowAPIAlways": true
         }
       }
     }

Landscape.uproject:

     {
	    "FileVersion": 3,
	    "EngineAssociation": "{1BCB251D-0C01-1204-0017-02C86FA949DF}",
	      "Category": "",
	       "Description": "",
	      "Modules": [
		{
			"Name": "Landscape",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": [
				"AirSim"
			]
		}
	    ],
	     "Plugins": [
		   {
			     "Name": "AirSim",
			     "Enabled": true
		    }
    	    ],
	      "EpicSampleNameHash": "1226740271"
}

anders-hopland, 2018-03-22:
I have had a similar issue. 
I first cloned the AirSim repo and built it using the following commands
`cd AirSim`
`./setup.sh`
`./build.sh`

After the plugin has been built, I copied the AirSim plugin located in `Airsim/Unreak/Plugins/` to the plugins folder in my project and deleted the old AirSim Plugin.

Try to run the UE4 editor from the command line to see output if anything goes wrong (this have helped me a lot). How to do this depends on where you installed Unreal Engine, but for me the command is `/home/<your-username>/UnrealEngine/Engine/Binaries/Linux/UE4Editor`.
When you then run your project, you will again be prompted with 
[image]
Here you choose no, and when you are asked if you want to rebuild you choose yes.

After all of this was done it worked for me, I hope the same applies for you

kdzy, 2018-03-24:
@anders-hopland 
Thanks  for your replay. I have followed your steps, but I still can't enable AirSim Plugin, and the error is as followed:
[image]

anders-hopland, 2018-03-24:
I'm not sure if this would work, but I first compiled the AirSim plugin from Github without any of my own code, and then added it to the project. This compiled just fine. After this, I copied all of my uasset files into my Airsim plugin, and rebuilt it from the UE4 editor. One problem I have faced is that I first built my project on Windows with Unreal Engine 4.16, but I need to use Unreal Engine 4.17 on Ubuntu, so some of my code is not functioning properly because of that. I have not been successful finding a solution to that problem yet.

I hope that this would work for you as well!

kdzy, 2018-03-25:
@anders-hopland 
Thanks. 
I want to know how to rebuild the project that the AirSim plugin and asset files have been copied to  from the UE4 Editor on Ubuntu.  Because I only compiled the AirSim and UE4 at the start without building project, so how to build project on Ubuntu?

anders-hopland, 2018-03-25:
@kdzy 
I'm not sure if I understand the problem. Have you managed to run the project from the UE4 editor, and need to rebuild e.g lighting. If this is the case, go to build in the top of the UE4 editor after you have opened the project. 
After you have copied over the AirSim plugin over to the project, you should only need to open it in the editor, close it, copy over the content to the content folder in the AirSim plugin which is created on first build in UE4 and then start it again from the UE4 editor.

kdzy, 2018-03-25:
@anders-hopland
I have tried your suggestions, but the same error still appears. Could you please tell me your setting up your environments on Ubuntu step by step in details if your are convenient.  I would appreciate  it if you can help me. Thank you a lot!

anders-hopland, 2018-03-26:
@kdzy 
Sorry for the late reply. I do not have access to the computer I used when i made it, but will gain access to it after easter break, but I think this should be about right.

- Download UE4 editor for ubuntu as follows (from the docs)
```bash
   # go to the folder where you clone GitHub projects
   git clone -b 4.17 https://github.com/EpicGames/UnrealEngine.git
   cd UnrealEngine
   # the Unreal build was broken a few times so we will get the commit that works
   git checkout af96417313a908b20621a443175ba91683c238c8
   ./Setup.sh
   ./GenerateProjectFiles.sh
   make
   ```
- `git clone https://github.com/Microsoft/AirSim.git`
- `cd AirSim`
- `./setup.sh`
- `./build.sh`
- Remove AirSim plugin from your unreal project
- `cp <path_to_airsim>/AirSim/Unreal/Airsim <plugins_folder_your_unreal_project>`
- Open UE4 editor and select your chosen project
- Dont disable AirSim when you are prompted for it, and choose rebuild after that.
- It should now build properly, you can after it have been built close the UE4 editor and add your own content files to the plugin

I would recommend using the command ´/home/<your-username>/UnrealEngine/Engine/Binaries/Linux/UE4Editor`to run the UE4 editor as you will get be able to get detailed error logs.

I hope this works for you. If anybody have a better way of doing it, I would be happy to know.

madratman, 2019-01-23:
closing as this was resolved. We'll update the docs with suggestions soon

closed 2019-01-23 by madratman