microsoft/AirSim #900: Segmentation does not capture SkeletalMeshActor

reichardtj, 2018-03-22:
The segmentation API is great. However, it seems one can only add segmentation IDs to static meshes, as far as I understand using `simGetSegmentationObjectID`. 

In the [initialization of AirSim](https://github.com/Microsoft/AirSim/blob/695d8acf51c515628288b369c8812025d7b96a1d/Unreal/Plugins/AirSim/Source/AirBlueprintLib.cpp#L340), there is only an iteration over `UStaticMeshComponent`. And it is only the components pertaining to this class that I can access using `simGetSegmentationObjectID` - Is this correct, so far?

However, according to the [Unreal API](http://api.unrealengine.com/INT/API/Runtime/Engine/Components/UMeshComponent/index.html), there are also `USkinnedMeshComponent` and in particular `USkeletalMeshComponent`. 

The latter would make great pedestrians, I believe. Currently, one cannot select them programmatically in the API for segmentation, though it's possible to assign the `CustomDepthStencilValue` manually of course. 

I guess one would have to change things [here](https://github.com/Microsoft/AirSim/blob/695d8acf51c515628288b369c8812025d7b96a1d/Unreal/Plugins/AirSim/Source/AirBlueprintLib.cpp#L323-L327)
adding another `if`:
```
if (mesh->SkeletalMesh())
            return std::string(TCHAR_TO_UTF8(*(mesh->SkeletalMesh()->GetName())));
```
and [here](https://github.com/Microsoft/AirSim/blob/695d8acf51c515628288b369c8812025d7b96a1d/Unreal/Plugins/AirSim/Source/AirBlueprintLib.cpp#L377) iterating over a different class, such as `USkinnedMeshComponent`.

What is the rationale behind the choosing which class of meshes to iterate over in the initialization? My guess is that iterating over other classes might give you more than 255 different IDs...

How would you suggest to go about making the assignment of segmentation IDs more flexible?

sytelus, 2018-03-23:
We actually iterated over UPrimitiveComponent previously. However in large environment there are lots of these and it takes too long to iterate over them. So rational was that we mostly care about things we in Unreal Editor's World Outliner and that's pretty much all of it usually UStaticMeshComponent. However you are absolutely correct that we should include USkeletalMeshComponent and USkinnedMeshComponent. The changes you have described are perfect. Another way might be to find common class to all 3 - if there exist one. But otherwise, may be doing 3 loops is also ok. Please feel free to send pull request.

reichardtj, 2018-03-29:
I've issued a [PR 913](https://github.com/Microsoft/AirSim/pull/913). It's not a perfect solution, but it does the job.