microsoft/AirSim #901: Clockspeed causes problems

Pierrolo, 2018-03-22:
Hello,

I am trying to use Airsim to apply RL to the drone.
My state input is the view from the front cam,
my action are a continuous vector : [+North , +East, +Down, Yaw_rate] w.r.t. the view of the drone.
I am using the python API.
But I am wasting a lot of time because of wall clock = simulation clock. I tried changing the Clockspeed setting in `setting.json` ; but even just at `"ClockSpeed": 2` the drone will become higly instable, it will shake a lot and eventually just fall on the ground on its own ...
Does anyone have a solution? 

My setup is Win10, i5 7300U, 2.60Ghz, 16go RAM (with a good old Intel(R) HD Graphics 620, but that shouldn't matter I guess)

Thank you for your help :)

Pierrolo, 2018-03-28:
Hi,

So I tried on a new computer with i7 and GTX 1050 (4go) and even clockspeed at 2.0 causes the drone to shake and eventually crash on the floor, no matter what the action was...

Does anyone had a similar problem? I could really use some help please...

Thanks for everything

sytelus, 2018-04-12:
I'm working on enhancing this feature and will soon post about the progress.

Pierrolo, 2018-04-12:
Hi,
Thanks ! :) In the meantime I managed to do somehting by following the instructions in Issue #660 proposed by @valtsblukis. By changing the PidController according to his patch and by changing `SimModeBase::getPhysicsLoopPeriod() ` as follow: 

`long long ASimModeBase::getPhysicsLoopPeriod()`
`{`
    `if (getSettings().usage_scenario == kUsageScenarioComputerVision)`
        `return 2500000LL;`
    `else`
        `return 250000LL;`
`}`

And by doing so I greatly reduced the instability, but there is stille quite a difference beetwen `"ClockSpeed" : 1` and `"ClockSpeed" : 10`

sytelus, 2018-04-13:
Just fixed this issue. On a powerful desktop, I can run quadcopter simulation now at 100X clock speed!

closed 2018-04-13 by sytelus

guillem74, 2018-04-13:
@sytelus It does not work for me...

New Neighborhood binaries show this when selecting quadrotor:

[image]

Pierrolo, 2018-04-13:
I tried building it, and it didn't work. It showed this as error:
```

"C:\Documents\AirSim_build_2\AirSim-master\AirSim.sln" (cible par défaut) (1) ->
"C:\Documents\AirSim_build_2\AirSim-master\DroneShell\DroneShell.vcxproj" (cible par défaut) (2) ->
"C:\Documents\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj" (cible par défaut) (3) ->
(ClCompile cible) ->
  c:\documents\airsim_build_2\airsim-master\airlib\include\common\clockbase.hpp(75): error C2220: avertissement considéré comme une erreur - aucun fichier 'object' généré [C:\Documen
ts\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj]
  c:\documents\airsim_build_2\airsim-master\airlib\include\common\clockbase.hpp(75): error C2220: avertissement considéré comme une erreur - aucun fichier 'object' généré [C:\Documen
ts\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj]
  c:\documents\airsim_build_2\airsim-master\airlib\include\common\clockbase.hpp(75): error C2220: avertissement considéré comme une erreur - aucun fichier 'object' généré [C:\Documen
ts\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj]
  c:\documents\airsim_build_2\airsim-master\airlib\include\common\clockbase.hpp(75): error C2220: avertissement considéré comme une erreur - aucun fichier 'object' généré [C:\Documen
ts\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj]
  c:\documents\airsim_build_2\airsim-master\airlib\include\common\clockbase.hpp(75): error C2220: avertissement considéré comme une erreur - aucun fichier 'object' généré [C:\Documen
ts\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj]
  c:\documents\airsim_build_2\airsim-master\airlib\include\common\clockbase.hpp(75): error C2220: avertissement considéré comme une erreur - aucun fichier 'object' généré [C:\Documen
ts\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj]
  c:\documents\airsim_build_2\airsim-master\airlib\include\common\clockbase.hpp(75): error C2220: avertissement considéré comme une erreur - aucun fichier 'object' généré [C:\Documen
ts\AirSim_build_2\AirSim-master\AirLib\AirLib.vcxproj]
```

Thnaks