microsoft/AirSim #902: Change position and FOV of cameras

baker-Xie, 2018-03-23:
(1) Is there a way to modify the position of the camera relative to the car?
(2) Can I set different FOV values to different cameras? Modify the json settings seems have no effect.

---- (Remove above before filing the issue) -----

sytelus, 2018-03-23:
Yes, you can change it here: https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Car/CarPawn.cpp#L125

For FOV, search for FOV_Degrees in https://github.com/Microsoft/AirSim/blob/master/docs/settings.md

baker-Xie, 2018-03-23:
@sytelus 
Thanks for your reply. But in your method, I must modify the source code and build it again. In this way, I cannot use the scenarios in this repo like city/neighborhood.  Is there a better solutions?

kshitijgoel007, 2018-03-26:
@sytelus Hey! Is the `FOV_Degrees` setting diagonal FOV, horizontal FOV or vertical FOV? It is a little confusing if a developer wants to set different x and y direction focal lengths etc. Thanks for putting this in settings.json directly, really helpful!

sytelus, 2018-03-28:
Only horz FOV supported.

sytelus, 2018-04-12:
@baker-Xie currently we are not focusing on making this configurable but you are welcome to contribute this feature.

closed 2018-04-12 by sytelus

dimikout3, 2019-10-02:
So, if the Horizontal FOV is set in the settings.json, what is the constant value of vertical FOV ?

madratman, 2019-10-02:
Vertical FoV can be calculated by the aspect ratio: vertical FoV = image_height / image_width * horizontal FoV
If you're trying to set up a specific cmos sensor with some desired image resolution and aspect ratio, you can always do an upper bound on the horizontal fov and image width, and the crop the image top and bottom if the desired vertical fov is less.

dimikout3, 2019-10-02:
What I am actually trying to do is to calculate each pixels angle from the middle of the image. Let me explain, lets say we have a horizontal fov_degrees=90 and width=576 pixels. The rightmost pixel (575) will have a relative angle of 45 degrees and the leftmost (pixel=0) a -45 degrees, horizontal displacement. Similar, for the vertical axis. Therefore, each pixel will correspond to a vertical and horizontal angle, relative to the middle of the image, eg if we have a 1024*576 image pixel[512,288] will have 0 degrees vertical angle and 0 degrees horizontal angle as well, pixel[0,0] will have -45 degrees horizontal relative angle and +45 vertical angle (if we consider a 90 degrees vertical field of view as well). Then, I want to combine the aforementioned angles with the information of a depth camera in order to create a 3D representations of what a camera sees ... The result may seem like a lidar with coloring. Something like that ... 
[image]

ivannson, 2020-01-15:
@dimikout3 This looks great! I'm trying to do a similar thing. Have you managed to get yours working?