microsoft/AirSim #910: Setting different FOV for different CameraIDs

kshitijgoel007, 2018-03-26:
Hi!

I am trying to set different FOV for front center and downward center cameras. However, it seems just using `CameraID` in Capture Settings in the settings.json is not working as expected. Is there a config file that allows this functionality or this has to be treated at the API level?

sytelus, 2018-04-12:
Please post your settings.json. It should work...

Domi47, 2018-04-30:
@sytelus so is it possible to set different FOV for different camera IDs? On settings doc https://github.com/Microsoft/AirSim/blob/master/docs/settings.md , it only shows that you can set FOV depending on image type. Maybe this issue is related to #993

jonyMarino, 2021-04-30:
@Domi47, a camera has an image type and FOV associated with it. Everything working in last version at least.

Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!

closed 2021-04-30 by jonyMarino