microsoft/AirSim #911: AirSimNH crashes after spawn quadrotor model

BrunoXL, 2018-03-26:
When trying to lainch `AirSimNH`  with quadrotor the application shows the error on the image below:
[image].

Sorry for posting images, but I don´t know where log files are genetated.

sytelus, 2018-03-28:
I haven't seen this at all. Looks like its happening in pak file? May be download is corrupted? The logs can found inside Saved folder.

BrunoXL, 2018-03-28:
@sytelus I download `AIrSimNH` again  and the error remains.
I`m including below part of the log file:

        LogWindows:Error: === Critical error: ===
    LogWindows:Error: 
    LogWindows:Error: Fatal error: [File:D:\Build\++UE4+Release4.16+Compile\Sync\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp] [Line: 798] 
    LogWindows:Error: Rendering thread exception:
    LogWindows:Error: Assertion failed: !Block.RawRequest && !Block.Processed && !Block.Raw && !Block.CPUWorkGraphEvent.GetReference() && !Block.ProcessedSize && !Block.RawSize && !Block.bCPUWorkIsComplete [File:D:\Build\++UE4+Release-4.16+Compile\Sync\Engine\Source\Runtime\PakFile\Private\IPlatformFilePak.cpp] [Line: 3013] 
    LogWindows:Error: 
    ]LogWindows:Error: 
    LogWindows:Error: 
    LogWindows:Error: KERNELBASE.dll!0x0000000038CC4008
    LogWindows:Error: AirSimNH.exe!FOutputDeviceWindowsError::Serialize()
    LogWindows:Error: AirSimNH.exe!FOutputDevice::Logf__VA()
    LogWindows:Error: AirSimNH.exe!FDebug::AssertFailed()
    LogWindows:Error: AirSimNH.exe!FPakAsyncReadFileHandle::StartBlock()
    LogWindows:Error: AirSimNH.exe!FPakAsyncReadFileHandle::ReadRequest()
    LogWindows:Error: AirSimNH.exe!FTexture2DResource::LoadMipData()
    LogWindows:Error: AirSimNH.exe!FTexture2DResource::UpdateMipCount()
    [2018.03.28-19.20.39:817][ 57]LogWindows:Error: AirSimNH.exe!TGraphTask<`FTexture2DResource::BeginUpdateMipCount'::`2'::EURCMacro_UpdateMipCountCommand>::ExecuteTask()
    LogWindows:Error: AirSimNH.exe!FNamedTaskThread::ProcessTasksNamedThread()
    LogWindows:Error: AirSimNH.exe!FNamedTaskThread::ProcessTasksUntilQuit()
    LogWindows:Error: AirSimNH.exe!RenderingThreadMain()
    LogWindows:Error: AirSimNH.exe!FRenderingThread::Run()
    LogWindows:Error: AirSimNH.exe!FRunnableThreadWin::Run()
    LogWindows:Error: AirSimNH.exe!FRunnableThreadWin::GuardedRun()
    LogWindows:Error: KERNEL32.DLL!0x000000003A8E1FE4
    LogWindows:Error: ntdll.dll!0x000000003C6AEFC1
    LogWindows:Error: ntdll.dll!0x000000003C6AEFC1
    LogWindows:Error: 
    LogWindows:Error: 
    LogWindows:Error: 
    LogWindows:Error: 
    LogExit: Executing StaticShutdownAfterError
    LogWindows: FPlatformMisc::RequestExit(1)

lovettchris, 2018-04-05:
What is your computer specs (OS version, graphics card, RAM ?)

BrunoXL, 2018-04-05:
OS: Win 10 x64
Graphic Card: Intel Graphics 520 / Geforce 940MX ( I configured to use Geforce card)
RAM: 8GB

sytelus, 2018-04-12:
Related: https://forums.unrealengine.com/unreal-engine/announcements-and-releases/118686-4-16-released/page12

This looks like UE4 issue. You might get better response at https://answers.unrealengine.com.

closed 2018-04-12 by sytelus