microsoft/AirSim #918: Bad Z velocity control

jfkeller, 2018-03-30:
I have been controlling the drone through the Python API using moveByVelocity. If I call moveByVelocity(0., 0., 0., 0.05) in a loop every 0.05 seconds, for around a minute the drone stays on the ground, then it rises high in the air, then begins descending and eventually returns to the ground and repeats. The following code produces the described results. The estimated and true velocities are identical, so I think it is a control problem and not a state estimation problem. Also, I noticed that the timestamp from AirSim and system time drift apart over time. I'm not sure if this is related or something separate.

```
import AirSimClient as asc
import time

if __name__ == '__main__':
    client = asc.MultirotorClient()
    client.confirmConnection()
    client.enableApiControl(True)
    
    while 1:
        client.moveByVelocity(0., 0., 0., 0.05)

        start = time.time()
        state = client.getMultirotorState()
        sys_time = time.time()
        airsim_time = float(state.timestamp)/1000000000.
        end = time.time()
        print('system time: ' + str(sys_time))
        print('AirSim time: ' + str(airsim_time))
        print('estimated velocity: ' + str(state.kinematics_estimated.linear_velocity))
        print('true velocity: ' + str(state.kinematics_true.linear_velocity) + '\n')
        elapsed = end - start
        
        time.sleep(0.05 - elapsed)
```

jfkeller, 2018-04-03:
I figured out the problem. In PidController.hpp there is this code that calculates the time dt since the last PID update:
```
float dt = clock_ == nullptr ? 1 : (static_cast<float>(clock_->millis()) - last_time_)* config_.time_scale;
```
The static_cast<float> is causing a loss of precision since clock->millis() and last_time_ are both uint64_t. Here are some example of clock->millis(), last_time_, and what numbers they get casted to as floats:
clock->millis()       last_time_          static_cast<float>(clock->millis())   static_cast<float>(last_time_)
1522772381699 1522772381696 1522772344832                             1522772344832
1522772381702 1522772381699 1522772344832                             1522772344832

Since the current and last time are getting cast to the same values dt is 0. In the code, dt must be above a threshold for the I and D terms to get applied. Eventually both the current and last time will not get cast to the same value, which causes dt to be large and the I term gets clipped to its max value.

The fix is to simply removed the statIc_cast<float>. The line should be:
```
float dt = clock_ == nullptr ? 1 : (clock_->millis() - last_time_)* config_.time_scale;
```

sytelus, 2018-04-12:
@jfkeller - This was a HARD nasty bug. It's impressive to see you tracking this down! Thanks.