microsoft/AirSim #920: setting.json does not change

simba155, 2018-04-02:
I'm a beginner about arisim.
 I want to use the computer vision mode and am trying to modify the settings.json.
 If I play my PROJECT of UE4 after changing, setting.json returns to original state like the photo below.
[image]

What should I do?? Please help me..

lovettchris, 2018-04-05:
I am not seeing this problem.  My Settings.json looks like this:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,
  "UsageScenario": "ComputerVision"
}
```
And PythonClient\cv_mode.py works.  Remember to answer "no" to the prompt asking for "Would you like to use car simulation? Choose no to use quadrotor simulation".  If this doesn't work for you try using a different text editor.  The only time AirSim resets the settings file is if it fails to load it entirely which might happen if your text editor is changing the text encoding on the file.

closed 2018-04-12 by sytelus

FireTime, 2018-05-25:
I was able to reproduce this with a student. There seems to be an issue with a Documents folder that is hosted by OneDrive. Temporally disabling OneDrive did not seem to fix the issue. Our workaround was to change the students documents back to the normal windows one for the remainder of the lab. I'm not sure if it's related but the windows language was set to Chinese. Whenever play was selected in the unreal engine the settings.json file was replaced with a default one. Setting the file to read only caused errors. I haven't had an opportunity to look at the code butI have a hunch that on run AirSim is unable to see the settings.jsop file in the redirected OneDrive documents folder and creates a new file as it thinks the file is missing. This overwrites the file settings causing defaults to be loaded on very run. @simba155 do you happen to be using OneDrive with a redirected Documents folder?

XueYuSky, 2019-03-08:
This is the second time I met the same problem. Last time, about one year ago, I met this proble, I solved it by initialize my OS ,Visual studio 2015 and UE4. But this time I don't want to do that ,it's really time consuming . I 'll  find solution to solve it .
But how ?

XueYuSky, 2019-03-12:
I resloved it!

damilolah, 2019-08-16:
@XueYuSky Hi 

I'm having a similar issue. How did you resolve it?

CSUN1997, 2022-01-27:
I also got this problem. For me, it is caused by OneDrive. I solved it by setting onedrive untrack the Documents folder.