microsoft/AirSim #923: How to set the starting location of the drone?

yousefab95, 2018-04-03:
I've been trying to find a way to set the starting location of the drone, regardless of where PlayerStart is located. I've read [here](https://github.com/Microsoft/AirSim/issues/276) that setSimPosition does that, but I can't seem to find any documentation for it.
For clarification, my team and I are building a mission planner where the user is able to set the home location on a map. We would like to take the coordinates the user chose for the home location and start the drone simulation from there.

mitchellspryn, 2018-04-05:
Not sure about the simSetPosition, but I've been using the following pseudocode to successfully set the position of the car:

```python
position = Vector3r(x,y,z)
heading = AirSimClientBase.toQuaternion(roll, pitch, yaw)
pose = Pose(position, heading)
car_client.simSetPose(pose, True)
```

It'd probably work for drones too.

yousefab95, 2018-04-22:
Thank you @mitchellspryn 
Would this code be placed before or after client.armDisarm?

I'm not quite sure where to put the code. I tried putting it in various places in my script but the simulator keeps crashing whenever client.simSetPose is called.

sytelus, 2018-04-25:
should be after armDiarm all.

showaykerker, 2018-06-24:
I modify this to new apis 
```
position = airsim.Vector3r(x,y,z)
heading = airsim.toQuaternion(roll, pitch, yaw)
pose = airsim.Pose(position, heading)
car_client.simSetPose(pose, True)
```
and I get an error that went:
> AttributeError: module 'airsim' has no attribute 'toQuaternion'

RVBldr, 2018-11-21:
Internal VSTF 7437

closed 2021-04-30 by jonyMarino