microsoft/AirSim #924: Cannot display 5 cameras on Linux

anders-hopland, 2018-04-03:
I have made a AirSim plugin which works fine on windows, but when I try to export it to Linux I have to rebuild the plugin. I did originally have 5 cameras, but when I use it on linux I only get 3. I have changed. I did originally change the following lines in `initializeSubWindows`:

```c++
//setup defaults
if (camera_count > 0) {
        subwindow_cameras_[0] = wrapper->getCamera(0);
        subwindow_cameras_[1] = wrapper->getCamera(camera_count > 1 ? 1 : 0);
        subwindow_cameras_[2] = wrapper->getCamera(camera_count > 2 ? 2 : 0);
	subwindow_cameras_[3] = wrapper->getCamera(camera_count > 3 ? 3 : 0);
	subwindow_cameras_[4] = wrapper->getCamera(camera_count > 4 ? 4 : 0);
}
else
subwindow_cameras_[0] = subwindow_cameras_[1] = subwindow_cameras_[2] = subwindow_cameras_[3] = subwindow_cameras_[4] = nullptr;

// Make all cameras capture scene images
subwindow_camera_types_[0] = ImageType::Scene;
subwindow_camera_types_[1] = ImageType::Scene;
subwindow_camera_types_[2] = ImageType::Scene;
subwindow_camera_types_[3] = ImageType::Scene;
subwindow_camera_types_[4] = ImageType::Scene;
subwindow_visible_[0] = subwindow_visible_[1] = subwindow_visible_[2] = subwindow_visible_[3] = subwindow_visible_[4] = true;
```

I also changed AirSimSettings.hpp:
```cpp
static constexpr int kSubwindowCount = 5; // Number of subwindows
```
This worked on windows, but did not work when I moved over to linux, subwindow 3 and 4 are just white. I created the subwindows in the UE4 editor, and copied over the AirSimHud uasset to the content folder in the plugin. 

Any ideas on this would be very much appreaciated!

sytelus, 2018-04-12:
This is pretty hard to debug without looking at code. I'd suggest cleanup your code, add configurability and send PR that folks can look at, try it out and provide you feedback.

closed 2018-04-12 by sytelus

zhuboshun2015, 2019-05-30:
@anders-hopland How do you add subwindows ？ Could you give me more details?