microsoft/AirSim #927: AirSim features

NextSim, 2018-04-03:
I'm working on a driving simulator that would connect AirSim to a power steering system (not a steering wheel joystick).

I have a few questions about AirSim:

- Can the car change between manual driving and autonomous driving on demand? Or does this have to be set up to one or the other before it is launched?

- How does autonomous driving work?  Is a path predefined and the car has to start at a defined location?  Can a destination be given and the car finds a route there?

- Is there are support for force feedback for the steering when going over curbs or bumpy roads?

NextSim, 2018-04-05:
Thanks.

Are there any details on the format of the API calls? I know there is the list of common API calls, but I haven't seen anything about the arguments it takes

mitchellspryn, 2018-04-06:
Not sure about formal documentation, but you can look through [AirsimClient.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/AirSimClient.py) to see what methods are available. For full examples, check out [HelloCar.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/hello_car.py).

NextSim, 2018-04-06:
Thanks.

I'm using c# but am struggling getting the API working.  I'm successfully connecting, my issue is sending the api data in the correct format.

I'm trying to use the serializer here:

https://github.com/neuecc/MessagePack-CSharp#extension-pointiformatterresolver

So to send enableApiControl I created a class

```
    [MessagePackObject(keyAsPropertyName: true)]
    public class enableApiHelper
    {
        [Key("enableApiControl")]
        public bool enable { get; set; }
    }
```
Create a class and set enable to true

```
                enableApiHelper apiEnable = new enableApiHelper
                {
                    enable = true
                };
```
Then convert it to a byte array and send it over the tcp connection (see data at the end)

I tried monitoring the data sent from the hellocar example in wireshark (that successfully enables the api and can control the car) and the data doesn't seem close.

I don't understand how to serialize the data properly into the MessagePack format.

I've been looking through the message pack serializer here as well, but I don't understand how to properly set the value/key pairs to serialize.

https://github.com/msgpack/msgpack-cli

Any help you can provide is greatly appreciated.. I'm really spinning my wheels.

0x81
0xa6
0x65
0x6e
0x61
0x62
0x6c
0x65
0xc3

NextSim, 2018-04-06:
I was able to figure out how to properly create the classes so that they serialize correctly.

Thanks.

sytelus, 2018-04-12:
Please continue discussion here: https://github.com/Microsoft/AirSim/issues/946

closed 2018-04-12 by sytelus