microsoft/AirSim #930: How can I record the depth image and the ground truth?

jinlukang1, 2018-04-04:
Hi!
I just started with AIrSim, but when I click the record, I found I can not get the depth image and the ground truth image but only the unprocessed image.
How can I record the depth image and the ground truth image?

Can you help me figure out what's wrong and what should I do?

jinlukang1, 2018-04-10:
Thanks for the reply,you have solved my problem,but how can I record the three types of images in the same time?

jinlukang1, 2018-04-10:
And I found that when I use the `"ImageType" : 5(as well as 3,4,6,7)` to record,I can only get one picture.
And this is my setting.json:
```
{
  "SettingsVersion": 1.0,
  "SimMode": "Car",
    "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
		  { "CameraID": 0, "ImageType": 5, "PixelsAsFloat": false, "Compress": true }
	  ]
}
}
```
Can you help me figure out what's wrong and what should I do?

mitchellspryn, 2018-04-10:
The "record" button only records one image at a time. To record more than one, use the python API. If you look around line 456 in [this source file](https://github.com/Microsoft/AutonomousDrivingCookbook/blob/master/DistributedRL/Share/scripts_downpour/app/distributed_agent.py) from our tutorial, you can see an example of how to use CarClient.simGetImages() to request an image as a numpy array. This API call takes in a list of ImageRequest, so you can request multiple images at the same time. Once you have them in numpy arrays, you can do whatever you wish with them, such as dumping them to disk or passing them to another processing function. Also note that if you try to record too many images, you may run into performance issues - rendering the images is somewhat expensive!

jinlukang1, 2018-04-11:
Thanks for your reply about how can I record the three types of images in the same time,it helps me a lot.

But I still feel puzzled about why I can only get one picture as my data when I choose 3 or 4 as my ImageType to record.

jinlukang1, 2018-04-12:
Sorry, my description may not be very clear.

When I record `"ImageType":1` and the `"RecordInterval": 0.05,`,I may get may 20 pictures in a second,but when I record `"ImageType":3` and the `"RecordInterval": 0.05,`,I just get one picture.

I don't know if it is reasonable,I thought I would get a lot of pictures in a few seconds.

sytelus, 2018-04-12:
Last description is not reproducible. You might want to verify if this is really happening.

closed 2018-04-12 by sytelus