microsoft/AirSim #931: Neighborhood map

guillem74, 2018-04-04:
In order to establish routes on my project I need an aerial image of the neighborhood map. 
As fas as I know, if you want to get it, you need to pay for the pack in Unreal...

Would someone be so kind to send me the image in a high resolution? Thanks

lovettchris, 2018-04-05:
Not sure this is what you want, but I published a highresshot of a top down looking view of the [neighborhood map](http://lovettsoftware.com/Downloads/Scratch/HighresScreenshot00000.png).   

This is 7680x4320px.  I used the built in command "highresshot 4" accessible from the game console which you get when you type ~ in a development version of the game.

mitchellspryn, 2018-04-05:
Also, if you're looking for raw coordinates of the roads, you can probably pull them from our Distributed Reinforcement Learning tutorial [here](https://github.com/Microsoft/AutonomousDrivingCookbook). The file you'd be looking for is /DistributedRL/Share/data/road_lines.txt, which would contain endpoints of each of the roads.

guillem74, 2018-04-05:
@lovettchris Thanks! Yesterday I collected pictures via manual mode of all the zone, similar to yours, so great.

@mitchellspryn Thanks! By now I am implementing drl methods with keras and openai gym to get from point a to point b and learn the routes. Then I will switch to multiple destinations routes (like similar to delivering packages to all the neighborhood). After that I will try to make lane detection. I'm also developing a UI web client with angular to interact with the python backend. This is all for my bachelor thesis degree in engineering. So, basically, now I'm checking for the opposite of road coordinates, I'm looking for each home coordinates :)

RVBldr, 2018-11-21:
Internal VSTF 7436.