microsoft/AirSim #932: Modified Vehicle Dynamics

mhdbabayi, 2018-04-04:
Hi 
Is it possible to change the Vehicle dynamics equations and use my own vehicle model?
I'm trying to use this for autonomous driving simulation and I need a more accurate vehicle model with more detail than NVIDIA PhysX.
I want to use my own equations of motion and force functions. 
Thanks

mitchellspryn, 2018-04-05:
I've done a bit of research into this. 

First, think really hard about whether you want to write your own physics engine. The physics of a car are extremely complex (for example, [look at how much work goes into modeling just the tires, and this is just curve-fitting](https://www.mathworks.com/help/physmod/sdl/ref/tireroadinteractionmagicformula.html)). Even the best simulators out there don't model vehicle behavior analytically; they rely on data from expensive tests on components to approximate vehicle behavior. I'd strongly recommend attempting to play around with the parameters inside the WheeledVehicle member of the SUV.uasset file. You have a surprising amount of control of the different parameters; you can get a wide variety of behavior from tweaking these values.

Also, consider that considerably more accurate engines already exist than stock PhysX. You may which to explore [torcs](https://sourceforge.net/projects/torcs/) or [bullet](http://bulletphysics.org/wordpress/), they might fit your need, and it would probably be less work to pipe data between these programs instead of writing them from scratch.

All that being said, AirSim already has an example of a custom physics engine. The multirotor part of the simulation actually doesn't use PhysX (for historical reasons - to my understanding, at the time AirSim was being initially developed, PhysX didn't update fast enough to allow for the fine-pitch motor control that was necessary for keeping the quadcopters in the air). If you look in AirLib/include/physics/FastPhysicsEngine.hpp, you should have an idea of how the forces are implemented. In short, UpdatePhysics() is called in a loop, after which your code would be responsible for detecting collisions and setting the Kinematics member of each of the physics bodies. For drones, the kinematics are relatively simple, as it's a free body with air resistance (and forces from the motors). For cars, this would be considerably more complex.

mhdbabayi, 2018-04-05:
Thanks a lot Mitchel 
We already have a validated and high fidelity vehicle model which we use for simulations. The problem is that model is in Matlab/Simulink and converting it to C++ for unreal engine takes some time. The reason I am interested in using Airsim is the visuals. We are trying to create a simulation platform which has high-fidelity in both the physics and the graphics so that it can be used by self-driving algorithms. 
I just wanted to make sure whether it is possible to have a different model solve the equations of motion  and update the states and use unreal engine just to get image data. 
Is there a tutorial available on how to setup a new physics engine, like the fast physics engine.
Thanks again for your help

mitchellspryn, 2018-04-06:
I see. There isn't a tutorial, but the work should be pretty straightforward - Create a class that inherits from PhysicsEngineBase, implement the UpdatePhysics() method which should pipe the current state of the vehicle / control signals into your matlab simulation and pipe the outputs of your simulation into the physics body, and select the engine in SimModeWorldBase.cpp. You can probably modify StandAlonePhysics.hpp as a quick test.

jeffreykxiao, 2018-04-06:
@mhdbabayi Have you tried:
https://jp.mathworks.com/matlabcentral/fileexchange/65966-vehicle-dynamics-blockset-interface-for-unreal-engine-4
https://www.unrealengine.com/marketplace/mathworks-interface

mhdbabayi, 2018-04-06:
Thanks a lot guys for the responses
I'll give these methods a try and let you know how it goes.

freeyyc, 2019-09-23:
> I see. There isn't a tutorial, but the work should be pretty straightforward - Create a class that inherits from PhysicsEngineBase, implement the UpdatePhysics() method which should pipe the current [...]

Is this method only for multirotor ?
If I want modify the physics engine for vehicle, How to do it?

Ischio, 2019-10-22:
> Thanks a lot guys for the responses
> I'll give these methods a try and let you know how it goes.

Did you managed to get something? I'm also trying to play with the environments dynamics.

thejeshk, 2020-02-21:
Hi @mhdbabayi @mitchellspryn 

How about this https://www.unrealengine.com/marketplace/en-US/product/carsim-vehicle-dynamics

I am also planning to use better vehicle dynamics but before that I thought of asking you guys if you have implemented anything better? 

Thanks and regards, 
Thejesh Kumar

mitchellspryn, 2020-02-21:
@thejeshk Do you have a copy of CarSim? You need that in order to use that plugin.