microsoft/AirSim #936: How to run HelloCar example

NextSim, 2018-04-04:
Can someone tell me the steps to run the helloCar example?

It's unclear to me what I need to run and in what order.

Thanks

closed 2018-04-04 by NextSim

had0909, 2018-04-26:
Excuse me, I have same problem. If you know how to work , can you share me ?

Thanks

NextSim, 2018-04-26:
Download and install a scenario at the link below (I used Africa) and run the run.bat file

https://github.com/Microsoft/AirSim/releases

Download the airsim project.  Run the Africa.exe and then run the hellocar project.

Qiaoyoufeng, 2018-08-07:
how to quit after you start  the exe, it is fullscreen, and i tried ESC and other key, seems not work, thanks

NextSim, 2018-08-07:
I always use alt + tab and then close the process.  We're working on adding an api for exit.  But as far as I know there isn't a way to just exit