microsoft/AirSim #937: AirSimClient python module cannot been found

BrunoXL, 2018-04-04:
How can I run `helloDrone` script from `PythonClient` folder?
I tried to add the code to my project but `IntelliSense` doesn´t find `AirSimClient` module and when I hit play button on Unreal Engine it doesn´t execute the standard code.

lovettchris, 2018-04-05:
The PythonClient code runs in separate process from the Unreal engine.  So you first launch your Unreal Environment containing the drone, then you can run "python hello_drone.py" from a python command line.  I use an [Anaconda ](https://docs.anaconda.com/anaconda/install/) command prompt.  The AirSimClient.py is in the same folder as hello_drone.py.  Once you have Python installed [VSCode ](https://code.visualstudio.com/) is also an excellent way to run and  debug python scripts.

BrunoXL, 2018-04-05:
@lovettchris thanks for reply. Now I´m able to run scripts, but I still got one issue. How can I run python scripts on my project? I have to copy and paste files to my project folders? Or there is a better way?

lovettchris, 2018-04-06:
Well with python the standard way to load modules from a different location is to append search paths to the sys.paths variable like this:
```python
import sys

# add the path to the folder that contains the AirSimClient module
sys.paths += [ "d:/git/Airsim/PythonClient" ]

# now this import should succeed
from AirSimClient import *
```

Now your script can live anywhere and it will be able to import the AirSimClient script from that path.  Is this what you are asking?

BrunoXL, 2018-04-07:
yes, thanks @lovettchris. Problem solved, issue closed.

closed 2018-04-07 by BrunoXL