microsoft/AirSim #938: Setting AirSim resolution in Windows

IAmSaad, 2018-04-05:
When I launch AirSim, it starts in full screen mode and it is set to max resolution of my monitor (1080p). How do I change the resolution to say 800*600 or less?

P.S. I am changing the rendering resolution because my GTX970 is unable to render at solid 30 fps in dense trees area in Neighbor Hood

lovettchris, 2018-04-06:
In the game folder you will find Saved\Config\WindowsNoEditor\GameUserSettings.ini.  Try changing this to the following:
```

[/Script/Engine.GameUserSettings]
 bUseVSync=False
 ResolutionSizeX=800
 ResolutionSizeY=600
 bUseDesktopResolutionForFullscreen=False
 FullscreenMode=2
 LastConfirmedFullscreenMode=2
 Version=5
```

IAmSaad, 2018-04-06:
@lovettchris There is no file named GameUserSettings.ini at the path provided. I made the file at the same path but AirSim still runs on full screen at 1080p

lovettchris, 2018-04-06:
very odd, it works for me.

IAmSaad, 2018-04-06:
I am using AirSimNH from release v1.1.8 on Windows

gusnitaa, 2018-04-09:
Try to start Airsim as follows : 
open the command prompt, go to the folder with airsim and execute in this way : 
for example : Airsim.exe -WINDOWED -ResX=640 -ResY=480
This should work.

IAmSaad, 2018-04-09:
Yes it works. Thanks.

cloudlakecho, 2020-05-23:
The INI file, `GameUserSettings.ini` is located under environmental name for example, if you have `Building_99` and then the INI would be at  `Building_99\Building_99\Saved\Config\WindowsNoEditor` folder