microsoft/AirSim #940: When I press backspace key, the wheels come off

kanchi0914, 2018-04-06:
Hi there, 

I was testing Built-in Blocks Environment but faced weird problem.
Specifically, while operating a car, pressing the backspace key causes the wheels to come of.
There was no problem in the binary sample environment, but in other environments I made, I faced the same problem.
I Attach the image when it happened.
[image]

Actually, moving the car is possible even if the tire runs out. But I want to explore the cause.
How do I fix it?

thanks.

sytelus, 2018-04-09:
That is super weird and have never seen this in any environments that we have. Is this reproducible? If so can you please give the steps?

sytelus, 2018-04-12:
We found repro steps:

1. Collide with something.
2. Press Backspace to do reset.